ExpressLRS/ExpressLRS #756: [3.0.1] CRSF transmit power on iNav OSD always shows 0mW

0crap, 2021-08-01:
## Current Behavior

As the title says, on all possible power steps I select, the OSD CRSF transmit power shows 0mW.

[image]

On my Taranis radio the TPWR sensor shows all power steps correctly.
(Besides 50mW which we know is a current limitation.)

From a Discord conversation I know this is a well known issue.
But with Dynamic Power around the corner it would be nice to have this one fixed.

So I made this issue just for the record, as a low priority fix.
Thx!

CapnBry, 2021-08-02:
This is because the RX doesn't know what power the TX is running at. It isn't so much a bug as it is a conscious choice to not waste bandwidth to send it to the RX.

Not that it wouldn't be nice, but it isn't an oversight, it is by design. Also, Betaflight doesn't have TX powerlevel in their OSD so maybe you can get them to add it for if we do implement this? (smile)

0crap, 2021-08-03:
Fair enough.
And a good choice not to waste any bandwith. I'm all in on that concept.
On Discord someone showed me it might be possible to sqeeze this info in.
Some guy named Capn something. ;-)

On BF I don't feel the need to show TX power.
Most of the long range stuff is done with planes (iNav), with Dynamic Power it would be nice to show TX power on the OSD.

Anyway if it seems not plausible to do on v1.1, please feel free to close this issue!

CapnBry, 2021-08-03:
I should have tagged this issue in #758, which has the TX power included as well. I was going to make a Betaflight PR for adding this but someone beat me to it, so the next BF version is going to have RF TX power in the OSD as well.

avsaase, 2021-09-04:
This was implemented in https://github.com/ExpressLRS/ExpressLRS/pull/821.

0crap, 2021-09-08:
Awesome!
I'll test this one and close it when this new (major) release will go out the door.

Feel free to close it now if needed for proper housekeeping. :-)

avsaase, 2021-09-08:
Nite that you need to enable the Wide switch mod for the transmit power value to be sent to the RX. That caught me out first.

CapnBry, 2021-11-04:
Fixed upstream in iNav

closed 2021-11-04 by CapnBry