ExpressLRS/ExpressLRS #764: R9 SLIM NO PLUS

Christianmix, 2021-08-05:
Hello everyone, I'm thinking about installing the express LRS on my R9M 2018, I have 5 RX R9 Slim, not being the plus, can the firmware used on the R9 Slim+ be used on my R9 Slim? Does firmware work for both?

StonedDawg, 2021-08-05:
@Christianmix you will need an ST-link tool for the R9 Slim with no "plus" and there is the target for it already. it's FRSKY_RX_R9SLIM_VIA_STLINK

Christianmix, 2021-08-05:
> @Christianmix você precisará de uma ferramenta ST-link para o R9 Slim sem "plus" e já existe um alvo para ela. é FRSKY_RX_R9SLIM_VIA_STLINK

ok, i can't find anything on slim, but i compiled the firmware with vscode, but there is a problem, watching videos about r9 mini, i saw that it is necessary to first install a temporary bootloader and then install the compiled firmware, in the case of r9 mini would be the file with this name: r9mm_elrs_bl.frk, ie a file with .FRK extension, and when I compile with vscode I have 2 files, firmware.bin and firmware.elf, where can I get the .FRK bootloader for the r9 slim?

Christianmix, 2021-08-07:
Flash error
Cloning into 'C:\Users\Christian Mix\.platformio\.cache\tmp\pkg-installing-gdm0mgsj'... .pio\libdeps\Frsky_RX_R9SLIM_via_BetaflightPassthrough\extEEPROM\extEEPROM.cpp: In member function 'byte extEEPROM::read(long unsigned int, byte*, unsigned int)': .pio\libdeps\Frsky_RX_R9SLIM_via_BetaflightPassthrough\extEEPROM\extEEPROM.cpp:185:51: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second: 185 | communication->requestFrom(ctrlByte, nRead); | ^ In file included from .pio\libdeps\Frsky_RX_R9SLIM_via_BetaflightPassthrough\extEEPROM/extEEPROM.h:62, from .pio\libdeps\Frsky_RX_R9SLIM_via_BetaflightPassthrough\extEEPROM\extEEPROM.cpp:58: C:\Users\Christian Mix\.platformio\packages\framework-arduinoststm32\libraries\Wire\src/Wire.h:100:13: note: candidate 1: 'uint8_t TwoWire::requestFrom(int, int)' 100 | uint8_t requestFrom(int, int); | ^~~~~~~~~~~ C:\Users\Christian Mix\.platformio\packages\framework-arduinoststm32\libraries\Wire\src/Wire.h:97:13: note: candidate 2: 'uint8_t TwoWire::requestFrom(uint8_t, uint8_t)' 97 | uint8_t requestFrom(uint8_t, uint8_t); | ^~~~~~~~~~~ *** [upload] Error -1 ========================= [FAILED] Took 61.58 seconds ========================= Environment Status Duration ----------------------------------------- -------- ------------ Frsky_RX_R9SLIM_via_BetaflightPassthrough FAILED 00:01:01.585 ==================== 1 failed, 0 succeeded in 00:01:01.585 ====================

Christianmix, 2021-08-07:
Processing Frsky_RX_R9SLIM_via_BetaflightPassthrough (platform: ststm32@8.0.0; board: bluepill_f103c8; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
UID bytes: 66,30,193,21,197,23
build flags: ['-D PLATFORM_STM32=1', '-O2', '-DTARGET_RX=1 -Wall -Iinclude', '-D TARGET_R9M_RX', '-D TARGET_R9SLIM_RX', '-D TARGET_100mW_MODULE', '-D HSE_VALUE=12000000U', '-DVECT_TAB_OFFSET=0x08008000U', '-DRegulatory_Domain_FCC_915', '-DMY_UID=66,30,193,21,197,23', '-DHYBRID_SWITCHES_8', '-DLOCK_ON_FIRST_CONNECTION', '-DFAST_SYNC', '-DLATEST_COMMIT=0x1,0xd,0x5,0x4,0x3,0xb']
Building for STM32 Platform
Building for SX1276 915FCC
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8.html
PLATFORM: ST STM32 (8.0.0) > BluePill F103C8
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200819 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - tool-dfuutil 1.9.200310 
 - tool-openocd 2.1000.200630 (10.0) 
 - tool-stm32duino 1.0.2 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <extEEPROM> 3.4.1+sha.4abc28d
|   |-- <Wire> 1.0
|-- <Wire> 1.0
|-- <SX127xDriver>
|   |-- <SPI> 1.0
|-- <SX1280Driver>
|   |-- <SPI> 1.0
|-- <FHSS>
|   |-- <SX127xDriver>
|   |   |-- <SPI> 1.0
|   |-- <SX1280Driver>
|   |   |-- <SPI> 1.0
|-- <POWERMGNT>
|   |-- <DAC>
|   |   |-- <Wire> 1.0
|   |   |-- <helpers>
|   |-- <SX127xDriver>
|   |   |-- <SPI> 1.0
|   |-- <SX1280Driver>
|   |   |-- <SPI> 1.0
|-- <EEPROM>
|   |-- <EEPROM> 2.0.1
|   |-- <extEEPROM> 3.4.1+sha.4abc28d
|   |   |-- <Wire> 1.0
|   |-- <Wire> 1.0
|-- <CRC>
|-- <CRSF>
|   |-- <CRC>
|   |-- <CrsfProtocol>
|   |-- <MSP>
|   |-- <TelemetryProtocol>
|   |-- <FIFO>
|-- <HWTIMER>
|-- <LQCALC>
|-- <MSP>
|-- <OTA>
|   |-- <CRSF>
|   |   |-- <CRC>
|   |   |-- <CrsfProtocol>
|   |   |-- <MSP>
|   |   |-- <TelemetryProtocol>
|   |   |-- <FIFO>
|-- <PFD>
|-- <STM32F3_WS2812B>
|-- <StubbornReceiver>
|-- <StubbornSender>
|-- <Telemetry>
|   |-- <CrsfProtocol>
|-- <TelemetryProtocol>
Building in release mode
PLATFORM : 'ststm32'
BUILD ENV: 'FRSKY_RX_R9SLIM_VIA_BETAFLIGHTPASSTHROUGH'
Checking size .pio\build\Frsky_RX_R9SLIM_via_BetaflightPassthrough\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  21.1% (used 4328 bytes from 20480 bytes)
Flash: [======    ]  56.1% (used 36748 bytes from 65536 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, custom, dfu, jlink, mbed, stlink
CURRENT: upload_protocol = custom
Uploading .pio\build\Frsky_RX_R9SLIM_via_BetaflightPassthrough\firmware.bin
=================== FIRMWARE UPLOAD ===================
  Bin file '.pio\build\Frsky_RX_R9SLIM_via_BetaflightPassthrough\firmware.bin'
  Port COM3 @ 420000
  Using CRSF (full duplex)!
======== PASSTHROUGH INIT ========
  Trying to initialize COM3 @ 420000
  Warning: 'No CLI available. Already in passthrough mode?, If this fails reboot FC and try again!'

Attempting to reboot into bootloader...
[1] retry...
[2] retry...
[3] retry...
[4] retry...
[5] retry...
[6] retry...
[7] retry...
[8] retry...
[9] retry...
[10] retry...
[FAILED] to get to BL in reasonable time
[FAILED] to get to BL in reasonable time

*** [upload] Error -1
========================= [FAILED] Took 30.95 seconds =========================

Environment                                Status    Duration
-----------------------------------------  --------  ------------
Frsky_RX_R9SLIM_via_BetaflightPassthrough  FAILED    00:00:30.953
==================== 1 failed, 0 succeeded in 00:00:30.953 ====================

pkendall64, 2021-08-07:
Please take this to discord help-and-support channel.

Christianmix, 2021-08-07:
> Por favor, leve isso para discordar do canal de ajuda e suporte.
where do i go to find this

pkendall64, 2021-08-07:
https://discord.gg/dS6ReFY and join discord and click on the help-and-support channel and someone there should be able to help you with getting this setup.

StonedDawg, 2021-08-13:
solved with guides from discord and link to :
https://www.expresslrs.org/quick-start/rx-stlink/

closed 2021-08-13 by StonedDawg