ExpressLRS/ExpressLRS #773: ELRS Lua script doesn't work on Frsky X Lite with R9M Lite

timemaster5, 2021-08-10:
## Current Behavior

elrs.lua still Loading 0/0 or Syntax error

## Steps to Reproduce

1. I have successfully flashed r9m lite module with ELRS FW. (bootloader first, then the FW according to wiki)
2. Install elrs.lua into SCRIPTS/TOOLS folder
3. Run it from Tools menu

## Possible Solution (Not obligatory)

No idea :(

## Details

I am on X-lite Pro, so took 128px grayscale version of lua script.
I have a model created for this, which uses an external transmitter and CRSF protocol.

I have tested on the latest OpenTX 2.3.14, then on 2.3.13 and also on EdgeTX 2.5.0, it is the same everywhere.

## Your Environment

- TX hardware: R9M lite module
- RX hardware: R9MM
- Handset model: FrSKY X lite Pro
- OpenTX version (including nightly number): stable 2.3.13, 2.3.14
- EdgeTX: 2.5.0 (f5dec4e3)
- ExpressLRS version (TX & RX MUST MATCH): compiled in one go 1.0.1
- ExpressLRS lua, from master today

pkendall64, 2021-08-10:
You need to use the matching LUA script. If you have 1.0.1 on the TX then you need the LUA from 1.0.1 branch.
Best to get it from the configurator app. There are major changes happening on master and ELRS.lua from master is not backward compatible.

timemaster5, 2021-08-10:
Ok, this was my stupid mistake, but leaving it here for the others :) @pkendall64 thank you very much, it was it. I just figured it out. The best way is really to use the version from the configurator. It works now, of course.

closed 2021-08-10 by timemaster5