ExpressLRS/ExpressLRS #774: Update error: ERROR[4]: Not Enough Space

miggy1952, 2021-08-11:
Using Configurator version Version 1.0.3 on a Mac using Safari and I get this message.

jurgelenas, 2021-08-11:
Can you verify that there is enough disk space in your computer?

miggy1952, 2021-08-11:
50GB available, should be enough?

jurgelenas, 2021-08-11:
Can you post full logs?

miggy1952, 2021-08-11:
Thanks for all you help but I am lost. How/where will I find the full logs?

miggy1952, 2021-08-11:
Ahhh! I should have mentioned that get the "ERROR[4]: Not Enough Space" when I am trying to flash EP Rx with wi-fi

fredbotton, 2021-08-11:
Just in case, did you double-check you did not try mistakenly to upload the TX binary to RX? It is bigger than the RX one, and I've ended with that Not enough space error trying to do it. I'm fortunate it did actually fail 8-)

miggy1952, 2021-08-11:
The firmware was called "firmware.elf" and is 1.5MB in size. Hope that helps?

fredbotton, 2021-08-11:
Seconded, that's firmware.bin. Around 850Kb for TX, 360Kb for RX.

miggy1952, 2021-08-12:
Yes that works! Many thanks all is good now.

closed 2021-08-12 by AlessandroAU