ExpressLRS/ExpressLRS #776: Lost link after fail safe, had to restart TX to regain it

maxbl4, 2021-08-11:
## Current Behavior

Was testing range of EP2 with SMD antenna. Tx at 250hz 100mw. Got several failsafes and after one of them, even when quad was much closer to myself, the link was not returning. Had to restart TX two times and regain control only when quad was around 70meters from me

## Steps to Reproduce

Induce failsafe several times, see if the link restores

## Details
The DVR and open tx logs can be found here https://disk.yandex.ru/d/jbER2GqahzYg5g also screenshot of configurator settings and RSSI chart is there as well

## Your Environment

- TX hardware: HappyModel ES24TX
- RX hardware: HappyModel EP2 RX
- Handset model: Jumper T-Lite
- OpenTX version (including nightly number): 2.3.14
- ExpressLRS version (TX & RX MUST MATCH): 1.0.1
- Packet Rate: 250
- Telemetry Ratio: 1:16
- user_defines: 
- -DRegulatory_Domain_ISM_2400
-DMY_BINDING_PHRASE="maxbl4"
-DHYBRID_SWITCHES_8
-DENABLE_TELEMETRY
-DLOCK_ON_FIRST_CONNECTION
-DAUTO_WIFI_ON_INTERVAL="20"
-DFAST_SYNC

maxbl4, 2021-08-11:
I have 100% repro. In short: if you have total loss of connection for about 10 seconds, you need to go back A LOT to regain it. Long version: set a quad on a bench, 500hz 10mw, go out of my house to around 20 meters. 1RSS -105, LQ 20-30. But still have connection. Cover tx antenna with hand for 1-2 seconds, loose connection. Open antenna again - connection restores. Cover tx antenna for 10-15 seconds - connection will not restore again. Walk back to my house, somewhere half way, like 10 meters, connection will restore. Looks like ELRS needs way way higher RSSI to restore a totally broken connection 🙁

JBKingdon, 2021-08-11:
yes. that's how it works. at those low rssi numbers you're out well brought the range at which it is possible to establish a link

maxbl4, 2021-08-12:
What RSSI is needed to establish a link?

maxbl4, 2021-08-12:
I did some more static tests. It's pretty easy with SMD antenna on 500hz 10mw, only need to go out of the house by 10-15 meters.
Example: RSSI -99, LQ 95. Drop link, wait for reconnection -no result. Gradually increase power to 25-50-100 - no result. Only setting 250mw resulted in link regain and RSSI of -85-87.
Also did some tests with 50hz, but's harder to test, it works quite fine at -100 RSSI and can regain link, but seem I'm just too close, need to get out by like 100 meters to get RSSI into comparable region with 500hz

avsaase, 2021-08-12:
> it's always likely to be the case that we can hold an existing link out beyond the range we can establish a new one.

Excuse my ignorance, but why is that? I can understand the tx and rx getting out of sync when the link is lost so it would take a while for the link to reestablish, and longer when fewer packets are getting through. Is there anything else that makes reestablishing a link difficult?

Having a very large difference between the LQ at which you failsafe and the LQ at which the link reconnects is kinda problematic. There are a lot of situations in which encounter noise that temporarily breaks your link but if it recovers quickly you can push through it. A friend of mine experienced this when he had a failsafe at 7km and the plane had to fly back 4km before the link recovered.

CapnBry, 2021-08-12:
I did extensive testing on this back when I dropped the required LQ from 75 to the current minlq of 5-9 (depending on regulatory domain) so I can speak to this with confidence.

There's not much LQ needed at all to establish a connection, and there no minimum RSSI required. The holdup is that you need to **receive** a sync packet at the RX. Sync packets are only sent one packet every >4 seconds. At low LQ, your chance of missing that packet is high, and therefore can take a long time to restart the connection process. Once the sync packet is received, the connection generally will return (switch from tentative to connected) very quickly. The main holdup is the infrequently sent, often missed sync packet. That may make it seem like you need a lot higher LQ, because your model is RTHing the whole time and every 4 seconds it has a higher chance to connect until the LQ at establishment is quite high.

I think it could help to increase the chances of the sync packet getting to the RX. The issue there is that the TX really doesn't know if the RX is receiving any packets since telemetry, especially at failsafe range, is almost never available. It is possible that increasing the sync packet frequency when telemetry STOPS being received would help (i.e. if telemetry _has_ been received, and it times out, then increase the sync rate). I'll give this a try in my low LQ test setup to see if it makes a difference.

The only real way to diagnose this is to hook up a debug UART and turn on the RX scoreboard to see what is preventing a reestablishment. I would say almost 100% of the time it is just the simple lack of a sync packet being received at all, and not any sort of implicit "high LQ" requirement.

maxbl4, 2021-08-12:
> @maxbl4 There's an important distinction between there being an RC link TX -> RX and the TX knowing there is an RC link via telemetry.

Yeah, you say, that I have TX->RX link, but without telemetry standing mere 20 meters away with -99 RSSI. I can even hear my quad beeping inside the house :))))
I'm running all tests on 10mw TX, and telemetry is 50mw

maxbl4, 2021-08-12:
After some more tests, knowing that i have to be very patient, i could regain link in around 2 minutes. So it works, just have to wait longer. Would be great if it's possible to improve this behaviour

CapnBry, 2021-08-12:
I've spent about 6 straight hours today connecting and disconnecting with various LQ and I can say with confidence that it is working as well as the design allows.

With an EP2 receiver the reconnect does seem to be worse than with the SIYI FR Mini RX. The SIYI has proper antennas and an incoming amplifier to I am guessing that helps as its reconnect time was almost always <4 seconds across at least 100 attempts at <75LQ. However, switching the sync channel to the middle of the spectrum also showed a noticeable improvement on the EP2, although I only did about 15-20 tests with each option. PeHo is going to be posting a PR with this improvement soon, which may be backported to 1.0. I'm also still pursuing increasing the sync interval after telemetry is lost but even the lower-bound there is pretty high (1.28s for 250Hz).

maxbl4, 2021-08-13:
I've ordered 30db attenuator, when I get it, I will be able to test ep1 receivers locally. For now I performed all tests on EP2 rx, maybe SMD antenna contributes to the issue

closed 2021-12-24 by maybenikhil