ExpressLRS/ExpressLRS #777: BETAFPV 915 TX   WIFI UPDATA CRASH

zhoumigo, 2021-08-12:
## Current Behavior

## Steps to Reproduce

1.BETAFPV 915 TX   WIFI UPDATA CRASH

2.CAN NOT USE WIFI UPDATA TO FLASH 

3.

4.

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware:
- RX hardware:
- Handset model:
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH):
- Packet Rate:
- Telemetry Ratio:
- user_defines:

pitts-mo, 2021-08-12:
This is unfortunate @zhoumigo :-/
@wvarty did do a PSA for some of the betafpv TX modules here:
https://www.youtube.com/watch?v=Pt5rMp7Zs8s

I am not sure if this applies to your 915mhz variant.
-p

wvarty, 2021-08-12:
Yep. This affects 2.4 and 900mhz BETAFPV modules unfortunately.
At the moment this doesn't appear to be a software crash, as the module also drops the "monitor" UART connection when wifi is enabled, indicating that the ESP32 is resetting due to some sort of hardware issue (power maybe, not sure). If the crash was software, we should see the exception into on the debug UART.

zhoumigo, 2021-08-18:
> 是的。不幸的是，这会影响 2.4 和 900mhz BETAFPV 模块。
> 目前，这似乎不是软件崩溃，因为在启用 wifi 时，模块还会断开“监视器”UART 连接，这表明 ESP32 由于某种硬件问题正在重置（可能是电源，不确定） . 如果崩溃是软件，我们应该在调试 UART 上看到异常。

thx so much! i get it!

zhoumigo, 2021-08-18:
> 这是不幸的@zhoumigo :-/
> @wvarty确实为这里的一些 betafpv TX 模块做了 PSA：[https](https://www.youtube.com/watch?v=Pt5rMp7Zs8s) : [//www.youtube.com/watch?v=](https://www.youtube.com/watch?v=Pt5rMp7Zs8s) Pt5rMp7Zs8s
> [](https://www.youtube.com/watch?v=Pt5rMp7Zs8s)
> 
> 我不确定这是否适用于您的 915mhz 变体。
> -p

thx bro ! i get it!

closed 2021-12-24 by maybenikhil