ExpressLRS/ExpressLRS #779: via_BetaflightPassthrough not working if the FC is flashed with INAV

miggy1952, 2021-08-12:
Works fine if the FC is running INAV (Matek 405 std) but did not work is the same fc is running INAV

P-I-Engineer, 2021-08-12:
what rx? it works with most.

miggy1952, 2021-08-13:
It was on a R9 MM-OTA. Worked fine when the F405 was flashed with Betaflight.

P-I-Engineer, 2021-08-13:
i'm using a ghost rx (stm based, same as r9) and have no issues flashing passthrough with inav.

try turning telemetry off in inav and see if it flashes.

aMax52, 2021-08-13:
Well these are little information we are getting here. I flashed my SIYS Mini on a F405 / INAV 2.61 with ELRS version 1.0.
 No problem. 
Are you on INAV 3.01? Keep in mind, if you are a Win 10 user, this can be even a problem , because sometimes this OS plays roulette with ports after an patch update.

CapnBry, 2021-08-14:
I also have no issues flashing though iNav on EP1 and EP2 RX, but have not tried my R9 Mini. There was a bug in older versions of iNav that broke the passthrough, which iNav are you on?

Also as P-I-Engineer suggested, maybe try with the Telemetry feature turned off in iNav, and also be sure your TX is off.

miggy1952, 2021-08-15:
Using INAV 3.01, I "Think" it might be related to having the FC powered by a LiPo or powered via the USB. Will get back to you with my results soon.

miggy1952, 2021-08-16:
False alarm! My apologies!!! All works fine in both INAV and Betaflight.

closed 2021-08-16 by miggy1952