ExpressLRS/ExpressLRS #780: Happymodel 2.4  EP1  Release 1.0.0     1.0.1  does not work

al1965al, 2021-08-13:
## Current Behavior
I have 3 EP1 receivers.
All 3 receivers were bought from one seller - a set.
Starting from version 1.0.0 RC9 - all receivers do not work. After uploading the firmware higher 1.0.0 RC9 - slow blinking.
No bind!  And no wifi (after 20 seconds).
Only slow blinking.
You can restore work only with firmware -UART ( CP2102) connecting two pads GND.) 1.0.0.RC9
The last two realizes 1.0.0 and 1.0.1- do not work.
Changed versions of configurators.The problem remains.
Last working 1.0.0 RC9
I flashed   3 RX - all ways. All 3 receivers work fine on firmware 1.0.0.RC9
Realizes 1.0.0 and 1.0.1- do not work.
This problem is on all 3 receivers. 1.0.0 RC9 works fine on all 3 receivers.
Realizes 1.0.0 and 1.0.1- do not work also on 3 receivers.
Hope the developers fix this problem.

## Steps to Reproduce

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware:
- RX hardware:
- Handset model:
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH):
- Packet Rate:
- Telemetry Ratio:
- user_defines:

SunjunKim, 2021-08-14:
Did you update your TX too?

al1965al, 2021-08-14:
TX was flashed and works fine on 1.0.0 and 1.0.1.
Problem with RX only.
Today I flew 11 km on wing.  25mW \ 10Hz EP1. Excellent result.
 TX was 1.0.1 and RX 1.0.0 RC9
Problem with RX only. Version 1.0.0 and 1.0.1 on RX -does not work.

SunjunKim, 2021-08-14:
> TX was flashed and works fine on 1.0.0 and 1.0.1.
> Problem with RX only.
> Today I flew 11 km on wing. 25mW \ 50Hz EP1. Excellent result.
> TX was 1.0.1 and RX 1.0.0 RC9
> Problem with RX only. Version 1.0.0 and 1.0.1 on RX -does not work (on my receivers).

As my HappyModel TX/RXs are working well on 1.0.0 & 1.0.1, there's no fundamental flaw on HW/SW.
Could you elaborate? Especially your firmware settings and your environments?

SunjunKim, 2021-08-14:
The changes between RC9 -> 1.0.0 were mainly adding BETAFPV target, and no practical change was applied on the HappyModel target. My top suspect is some unmatching configuration when you build your 1.0.x RX firmware.

CapnBry, 2021-08-14:
My HappyModel EP1 works fine on 1.0.0 and 1.0.1. I can't think of what would make them stop working. Maybe you can tell us specifically which git revision makes it stop working? Also please fill in this information in the first post (especially user defines)
```
TX hardware:
RX hardware:
Handset model:
OpenTX version (including nightly number)
ExpressLRS version (TX & RX MUST MATCH):
Packet Rate:
Telemetry Ratio:
user_defines:
```

al1965al, 2021-08-14:
Taranis qx7. First version. Without mod. Computer -64bit. Windows 7.Configurator settings are default.Just added a phrase for bind.
ISM-2400

al1965al, 2021-08-14:
_My top suspect is some unmatching configuration when you build your 1.0.x RX firmware._
This is not my first day at the computer! And I flashed the firmware 20 times!!!

SunjunKim, 2021-08-14:
> _My top suspect is some unmatching configuration when you build your 1.0.x RX firmware._
> This is not my first day at the computer! And I flashed the firmware 20 times!!!

Well.. even experts make mistakes. No offense. Just let us double-check everything is right.
So could you please kindly upload your user-defines, both the TX/RXs?

AlessandroAU, 2021-08-14:
There is barely any difference between RC9 and release, just a few extra hardware targets. It doesn't make sense RC9 works and release version does not :|

al1965al, 2021-08-14:
Other people also have this bug! It was they who advised to immediately install 1.0.0.RC9.
I'm not alone.

SunjunKim, 2021-08-14:
> Other people also have this bug! It was they who advised to immediately install 1.0.0.RC9.
> I'm not alone.

I believe you. However to find the problem we need your user-defines settings, or the screenshots of your configurator settings.

al1965al, 2021-08-14:
No problem. I will made a screenshot and send it to you by email. Where can I send it?

SunjunKim, 2021-08-14:
https://imgur.com/ upload here and link them

al1965al, 2021-08-14:
Ok.Thanks.
Understand me. I am very pleased with the ELRS link. VERY! Excellent range and overall performance!!
But there is a fact - for me 1.0.0 and 1.0.1 - does not work on RX!

al1965al, 2021-08-14:
you can see it?
https://c.radikal.ru/c26/2108/47/be5f933921f9.jpg

SunjunKim, 2021-08-14:
Your TX settings as well?

al1965al, 2021-08-14:
No problem with TX! All versions TX  work EXCELLENT!
I will say more.
I also have R9M -915.
Everything works fine. TX and RX! All versions! (1.0.0 and 1.0.1)
Bug only with RX Happymodel 2.4

al1965al, 2021-08-14:
Of course the transmitter has exactly the same settings!
The configurator sets them up automatically.

SunjunKim, 2021-08-14:
We need more information. I (and probably devs as well) am confused because:
1. There's virtually no difference between RC9 and 1.0.0 on HappyModel target
2. My HappyModel RXs are working without any problem on 1.0.x
3. I cannot reproduce your problem.

Please describe the procedure to reproduce your problem, with as much detail as possible. Please include even the smallest piece of information, even though you think that's an obvious thing and does not need to be described.

More information you give, more chance we find the problem.

al1965al, 2021-08-14:
We turn on the brains!
Configurator settings are NOT IMPORTANT at all! The transmitter settings are also not important.
Even if the transmitter is turned off - the receiver should blink quickly - waiting for the bind!And then transition to Wi-Fi
My receivers are blinking slowly.

SunjunKim, 2021-08-14:
How slow? slower than this?
https://www.youtube.com/watch?v=ISYJaVLxpj4&lc=UgxqxeVEyfQD8x2suXV4AaABAg

al1965al, 2021-08-14:
No. This is when the receiver is working normally. I have it slowly.

SunjunKim, 2021-08-14:
> I have it slowly.
Do you mean your receiver is blinking slower than the video I posted?

al1965al, 2021-08-14:
Yes, slower.

al1965al, 2021-08-14:
Yes, slower.

SunjunKim, 2021-08-14:
Your tested configurator versions?
And of course, you've applied the same configurator settings on the RC9 and 1.0.0 versions, right?

al1965al, 2021-08-14:
Everything is correct. 
I checked different versions of configurators. The problem is not given by the configurator. Only the firmware.

al1965al, 2021-08-14:
Different configurators - the problem remains.

al1965al, 2021-08-14:
It does not matter what settings in the configurator. The receiver EVEN without a transmitter should blink faster while waiting for the bind .. I have it blinking slowly.

SunjunKim, 2021-08-14:
> Different configurators - the problem remains.
When you say something about the information you only know and not consistent for everyone, COULD YOU PROVIDE  MORE DETAILS? like version numbers, etc?
In this case, list all the configurator versions you've tested will be a good instance.

al1965al, 2021-08-14:
Tomorrow I will try to make a video of how a normally working receiver blinks - and how it blinks which does not work.

al1965al, 2021-08-14:
I cannot test all versions. I changed the versions of the configurator - and the problem remained. On any version configurator, version 1.0.0RC9 - works. And 1.0.0 - does not work.

SunjunKim, 2021-08-14:
https://github.com/ExpressLRS/ExpressLRS/issues/720

It seems like kinda same problem with this.

al1965al, 2021-08-14:
When I flash through the UART firmware ( CP2102) - e configurator always writes - the firmware was successful!
I turn off the power and then the RX slowly blinks.

al1965al, 2021-08-14:
The problem is not with the hardware. 
And not with the settings. I repeat.
I also have R9M -915.
Everything works fine. TX and RX! All versions! (1.0.0 and 1.0.1)
Bug only with RX Happymodel 2.4

CapnBry, 2021-08-14:
If you have a CP2102 hooked up to it, then you should open the serial monitor (420000 baud) after you flash and see what it says. It would be best to open the serial monitor first, then power the RX so you see the full log.

AlessandroAU, 2021-08-14:
@al1965al ,

Can you try something, in the configurator, click support, then clear platform files. Then rebuild and flash, it will force fresh downloads.

al1965al, 2021-08-14:
I repeat-UART firmware ( CP2102) - configurator always writes - the firmware was successful!
AlessandroAU
I'll try to clear the platform files - and download everything again.

al1965al, 2021-08-14:
Tomorrow I will try to clear everything and download it again.

CapnBry, 2021-08-14:
> I repeat-UART firmware ( CP2102) - configurator always writes - the firmware was successful!

If this is in response to me, that's not what I am saying. **After** you flash the RX with the CP2102, open the serial monitor to see the RX's operating log, not the log of flashing the RX.

al1965al, 2021-08-14:
I understood.
I'll look at the firmware log.

al1965al, 2021-08-14:
Thanks everyone.

al1965al, 2021-08-15:
2 of my receivers. The voltage is supplied to both receivers at once. I just flashed one via UART firmware (( CP2102)for version 1.0.1 (does not work)
Another with version 1.0.0.RC9.
See the difference in blinking speed.
https://www.youtube.com/watch?v=G8dafYWMsXM

The settings in the configurator are not important at all.
It blinks incorrectly.
Then I changed the version from 1.0.1 to 1.0.0RC9 on the same computer. And I also flashed RX with all the previous settings - and the receiver worked as it should.

al1965al, 2021-08-15:
After installing the UART 1.0.1 firmware (which does not work), I cut out the log file from the log folder (the file was the only one in the folder).
And I can send it by email.
Give me an email.

Sekilsgs2, 2021-08-15:
Maybe problem because new releases using sdk 3.0 for esp and something wrong in PlatformIO local sources and need cleanup old files or clean reinstall PlatformIO with new sdk 3.0... 
Maybe need trying  - 
```
Delete this folder C:\Users\UserName\.platformio\platforms\espressif8266@src-?????????????
Delete .pio and .vscode folders in your project.
Clean and Compile the project.
```

al1965al, 2021-08-15:
I repeat!
According to Alexandro's recommendation, I first deleted the files in the configurator and then deleted the entire configurator from Windows.
Then I reinstalled the configurator.1.0.3
I have two computers and both have the same problem.

Sekilsgs2, 2021-08-15:
> I repeat!
> According to Alexandro's recommendation, I first deleted the files in the configurator and then deleted the entire configurator from Windows.
> Then I reinstalled the configurator.1.0.3

Configurator dont delete this files! Configurator and PlatformIO its two different systems and configurator installing if needed PlatformIO and modules..
Try deleting entire folder C:\Users\UserName\.platformio\platforms\espressif8266 and try again and reply.
And if other peoples have working EP1 with latest releases than this 100% mean that this problem in your side. good luck.

al1965al, 2021-08-15:
I'll try. Thanks.

al1965al, 2021-08-15:
In the folder - platforms -I have 
espressif32
espressif8266
espressif8266@2.6.3
ststm32
What i need delete?

Sekilsgs2, 2021-08-15:
> espressif8266
> espressif8266@2.6.3

al1965al, 2021-08-15:
Deleted folders. Cleaned files. Sorry, the problem persists. Slow blinking.
I flashed again 1.0.0 RC9 works fine.

Sekilsgs2, 2021-08-15:
> Deleted folders. Cleaned files. Sorry, the problem persists. Slow blinking.
> I flashed again 1.0.0 RC9 works fine.

Ok thank. need help from dev :)

CapnBry, 2021-08-15:
That short flash doesn't look like anything we intentionally produce, so I think it is actually just rebooting itself over and over. Can you post the Serial Monitor stuff here? It should look like this on a working RX:
```
ExpressLRS Module Booting...
Setting 2.4GHz Mode
Setting 2400MHz Mode
Number of FHSS frequencies =80
0 65 68 44 55 57 20 37 63 3
64 28 48 4 61 32 69 13 75 52
79 47 45 18 49 30 29 15 73 43
27 6 26 74 39 50 12 9 66 54
56 8 78 11 51 25 21 72 71 76
58 40 42 1 36 41 70 35 38 23
67 2 46 16 22 5 14 62 77 53
33 7 17 31 34 60 19 59 24 10
0 76 70 15 46 14 58 41 56 22
28 4 19 12 29 8 2 65 42 71
27 74 9 60 1 73 67 30 39 69
36 25 18 53 79 43 64 55 11 66
37 75 17 72 61 78 40 49 34 35
68 13 38 54 16 63 7 45 57 5
44 59 20 62 10 6 31 47 21 33
3 23 52 32 77 24 26 50 48 51
0 54 56 3 19 32 4 71 17 65
34 29 43 59 18 35 53 61 48 26
57 50 20 21 46 42 27 55 7 67
70 16 58 40 72 51 49 15 78 38
39 22 64 30 74 60 68 47 52 62
63 13 73 66 45 69 77 23 76 10
33 36 25 6 11 44 28 41 24 75
14 2 79 31 5 8 12 37 1 9
0 18 79 33 74 38 46 39 60 51
13 31 22 26 53 24
Hal Init
Use TX pin: 30
Config SPI
SX1280 Reset
SX1280 Ready!
SX1280 Begin
Read Vers: 43447
SetPower: 11
SetPower: 17
About to start CRSF task...
2000
4000
6666
20000
```

I'm looking at our commit log and just not seeing anything from RC9 to 1.0.0 that should make any difference. Have you also tried `master` branch just to see if it does the same blink pattern?

estebanavv, 2021-08-28:
I have 2 EP1 RXs failing in similar way they have erratic behavior, sometimes they are in boot mode but most of the times when power is connected it does a quick flash and led becomes dim and dos the quick flash every 5 to 10 secs. Both where succesfuly flashed v1.0.1 via FTDI and configurator 1.0.3.

I am not experienced enough to debug it but I am willing to mail it to anyone in the US willing to debug it.

estebanavv, 2021-09-03:
Based on the discord chat I upgraded to 1.0x Maintenance. Now both receivers are working without issues.  https://github.com/ExpressLRS/ExpressLRS/commit/4a1b8e459511fc10542e2e0253997202e9155f3b

al1965al, 2021-09-04:
I have no experience with Arduino.
Maybe the developers will still make the corrected firmware?
It is already clear that the problem is not only with me.And many other people.

pkendall64, 2021-09-04:
> I have no experience with Arduino.
> Maybe the developers will still make the corrected firmware?
> It is already clear that the problem is not only with me.And many other people.

There is a fix in the 1.0.x-maintenance branch. Please flash that branch and report back.

al1965al, 2021-09-04:
Version 1.0.1 did not work. There was no new version after 1.0.1. Where is it fixed?
Maybe I should try with the updated configurator?

pkendall64, 2021-09-04:
> Version 1.0.1 did not work. There was no new version after 1.0.1. Where is it fixed?
> Maybe I should try with the updated configurator?

In configurator you need to choose the "Git Branch" tab and select "1.0.x-maintenance" as the branch to build.

al1965al, 2021-09-04:
Or something was fixed in old 1.0.x ?

[2 more replies]

closed 2021-09-08 by al1965al