ExpressLRS/ExpressLRS #788: Discussion - Remove ELRS_TELEMETRY_TYPE_LINK and make ELRS_TELEMETRY_TYPE_DATA permanently enabled

JyeSmith, 2021-08-16:
What do we think about removing the basic tlm packet type ELRS_TELEMETRY_TYPE_LINK and just having ELRS_TELEMETRY_TYPE_DATA permanently enabled.  If users wish to only have basic tlm then unwanted sensors can be disabled within BF etc.

schugabe, 2021-08-16:
I think this would be a good way forward to merge "basic/advanced". The place to disable sensors is on the FC side because we cannot support every possible configuration that somebody would like to have in basic tlm. I see the telemetry support as a "dumb" transportation layer that simply transmits all data received from the FC. Otherwise there is always the risk that some special sauce code for one FC firmware does not work for other variants and makes the support for this feature more complicated.

One open question would be if the link message still should be sent with a higher priority: if this is still required I'd say we can keep the two different ELRS_TELEMETRY_TYPE_LINK and ELRS_TELEMETRY_TYPE_DATA but always compile with the ENABLE_TELEMETRY and move tlm to a config option that can be enabled/disabled via the lua script (or just continue to use the rate setting) and not a compile time setting. This would allow to use the bf lua script if necessary without the need to flash rx/tx first.

CapnBry, 2021-08-16:
I can see just making ENABLE_TELEMETRY the only way, but I think the LinkStats packet should stay. It is a telemetry packet that is always there even if we're not connected to a FC [properly]. It is already throttled in how often it gets sent (2x a second) so what's the advantage of encapsulating it as advanced telemetry which would cause it to take more bandwidth than it does now on a single packet?

What is the goal of removing the LinkStatistics packet?

Billbo911, 2021-08-16:
To follow Jye's logic, leaving ELRS_TELEMETRY_TYPE_DATA permanently enabled would allow BF users to trim out all unwanted DATA and only receive what is needed for their flying style. That said, it also works for iNav pilots. iNav does not have the ability to remove specific TM Data (currently), but iNav user usually want more TM DATA anyway. So, this makes having ELRS_TELEMETRY_TYPE_DATA permanently enabled a win win for both iNav and BF pilots.

closed 2021-08-22 by JyeSmith