ExpressLRS/ExpressLRS #791: AUX1 arming for arming-less vehicles

padcom, 2021-08-16:
Currently, most of the documentation when it comes to channel allocation states to put arming on channel AUX1, because of its obvious low latency and priority (being sent with every packet and all). This is great and very desirable for quads, but pointless for other types of vehicles, like planes, rovers and boats where stick arming is very popular.

I understand that the idea behind the AUX1 being a binary, low latency switch is to make sure the command always gets to the receiver and later on to the flight controller. This is actually great news for long range pilots where the RTH command is indeed the last resort once all the video link is gone. Without it operating 100% reliably, it is easy to get the plane to fly further away and risking that the battery will run out. Having RTH on channels AUX2-8 has a 1-in-7 chance of being missed (as per current documentation).

I also completely understand that there is this additional magic going on with lowest-latency (experimental FEATURE_OPENTX_SYNC feature) that uses that information about the vehicle being armed/disarmed to speed up the transmission by eliminating some of the transmitted data. That's, again, great for ultra-low-latency applications, but completely useless for mid/long range flying where the link is expected to be lost.

I think it would be great to revisit the docs and take other types of vehicles into account.

padcom, 2021-08-16:
I think the bottom line is that arming is crucial for multirotors whereas return to home is crucial for long range flying with a fixed wing and hence the AUX1 should have proper function based on the application.

SunjunKim, 2021-08-19:
You can always use the AUX1 switch for the RTH function. It's not hard-coded for the "Arm" function exclusively. It will have all the benefits of the prioritized information for RTH triggering.
* make sure you disable NO_SYNC_ON_ARM in this case.

0crap, 2021-08-23:
> You can always use the AUX1 switch for the RTH function. It's not hard-coded for the "Arm" function exclusively (except on NO_SYNC_ON_ARM feature). It will have all the benefits of the prioritized information for RTH triggering.
> ps. make sure you disable NO_SYNC_ON_ARM in this case.

^Exactly this.

Feel free to use channel 5 for anything you like if you understand the risks.
Nothing needs to be changed here.

closed 2022-01-12 by maybenikhil

avsaase, 2022-01-13:
> You can always use the AUX1 switch for the RTH function. It's not hard-coded for the "Arm" function exclusively (except on NO_SYNC_ON_ARM feature). It will have all the benefits of the prioritized information for RTH triggering. ps. make sure you disable NO_SYNC_ON_ARM in this case.

In case someone stumbles upon this issue in the future: as of 2.0, dynamic power also depends on the arming switch. When channel 5 is low and the transmitter does no receive telemetry, it will not increase the transmit power to the max configured level.

padcom, 2022-01-14:
@avsaase Can we somehow disable that AUX1 check?