ExpressLRS/ExpressLRS #827: KissCF.lua (based on bf.lua) not working.

sandrosov85, 2021-08-23:
## Current Behavior

Not working KissCF.lua from RC that was made for configuring kissfc.
KissCF.lua is based on BF.lua but not working with elrs. BF.lua - works fine.

On the display nothing happens... All values is " --- "

Already add [bugreport](https://github.com/flyduino/kissfc-tx-lua-scripts/issues/53).
Can you please check what is wrong with lua?

@fedorcomander already purchase elrs module and wait for delivery. 
If @ELRSTeam have some time please check script and help with edit or debug or send link for protocol documentation.

in this [comment](https://github.com/flyduino/kissfc-tx-lua-scripts/issues/53#issuecomment-841678859) @fedorcomander says about "maybe addresses are not ok". 

Please help!

## Steps to Reproduce

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware: Happymodel ES900TX
- RX hardware: FRSKY R9Slim+
- Handset model: Radiomaster TX16S
- OpenTX version (including nightly number): 2.3.14 (master)
- ExpressLRS version (TX & RX MUST MATCH): 1.0.1 from master (RX & TX version match)
- Packet Rate: 200HZ
- Telemetry Ratio: 1:16
- user_defines: 
REGULATORY_DOMAIN_FCC_915
UNLOCK_HIGHER_POWER
UART_INVERTED
FEATURE_OPENTX_SYNC
HYBRID_SWITCHES_8
ENABLE_TELEMETRY

schugabe, 2021-09-02:
@sandrosov85 What KISS FC do you use?

fedorcomander, 2021-09-02:
DOESNT MATTER… LRS __DOES NOT_ use that part of crossfire protocol at all.

schugabe, 2021-09-02:
What parts of crsf are you talking about? We can add the remaining parts to elrs.

schugabe, 2021-09-02:
So 0x78 sends the request from lua => FC and 0x79 is the response from the FC => lua?

What is the max size of the two message types?

fedorcomander, 2021-09-02:
Phoe, all is in kiss lua script, its open source, it was like 4 years ago mate. Size is the same for all crossfire packages.

From PDF

Chunks

Maximum frame size is 62 bytes ( 64 - header ).

schugabe, 2021-09-22:
@fedorcomander I added both frame types to elrs but I don't get any stick movement if I connect the elrs rx to a kiss v1. I use the same connections/settings as a crossfire rx would use. Do you have any idea what's missing here? I heard that some people run elrs with a kiss v2 but nobody with a v1 seems to have got it working.

fedorcomander, 2021-09-22:
hm. should work oob. 

---
Cheers,
Alex