ExpressLRS/ExpressLRS #831: R9 SX support

pskmac, 2021-08-24:
Hey - I did not find R9 SX which I own, on the supported TX list.
Has anybody actually tried it? R9 SX seems to be an slightly upgraded version of R9 Slim+ which is supported, so i just thought that  it might work on R9 SX too some way.

Thanks, Maciej

maybenikhil, 2021-08-24:
Currently no native recievers have PWM support

CapnBry, 2021-08-28:
Any flight controller that supports CRSF protocol and can output to servos will output to servos. We just send channel data.

pskmac, 2021-08-28:
> Any flight controller that supports CRSF protocol and can output to servos will output to servos. We just send channel data.

I apologize, I am not sure that I fully follow. As guys were mentioning about pwm previously I thought they meant direct connection of servo to rx. I'm using FC so will elrs work on r9 sx?

CapnBry, 2021-08-28:
No there is no support for FrSky R9 SX and there are no plans to support it since none of the developers has that hardware.

pskmac, 2021-08-28:
Thanks, clear to me now.

closed 2021-08-28 by pskmac

drtrigon, 2022-04-16:
> No there is no support for FrSky R9 SX and there are no plans to support it since none of the developers has that hardware.

I have one to spend if anybody is intressted just contact me.

pskmac, 2022-04-19:
> > No there is no support for FrSky R9 SX and there are no plans to support it since none of the developers has that hardware.
> 
> I have one to spend if anybody is intressted just contact me.

Me too ;-)