ExpressLRS/ExpressLRS #867: Ghost TX OpenTX Flashing Issues

maybenikhil, 2021-09-03:
## Current Behavior

Ghost TX LED freezes on blue when I attempt to flash it through openTX

## Steps to Reproduce

Compile latest master via ExpressLRS Configurator and press "flash external module" in EdgeTX.

## Details

The TX works fine when flashed via STLinkv2

## Your Environment

- TX hardware: IMRC Ghost TX
- RX hardware: N/A
- Handset model: OpenTX
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH): any master ver.
- Packet Rate: N/A
- Telemetry Ratio: N/A
- user_defines: N/A

Also @brandonrc asked me to tag @cruwaller

brandonrc, 2021-09-03:
@cruwaller 

I am going to test this out but looks like something happened with the bootloader on the master branch for ghost tx's.

cruwaller, 2021-09-03:
Ok, need to check. Have you flashed it successfully earlier?
Maybe it is something related to bootloader 🤔

brandonrc, 2021-10-05:
I have not seen this issue. 

was it resolved? @cruwaller @maybenikhil 

Just flashed my stuff to 1.0 to 1.1 to 1.0 and back to 1.1-oled

cruwaller, 2021-10-05:
No, I have done nothing and nothing is changed to bootloader.
@brandonrc , do you have the latest bootloader in use?

closed 2021-12-20 by maybenikhil