ExpressLRS/ExpressLRS #873: Telemetry packets dropped on 150Hz

avsaase, 2021-09-05:
## Current Behavior

There seems to be a problem with telemetry at 150Hz. On the `master` branch, the `connectionState` is now set to `disconnected` when 5 consecutive telemetry packets are missed (the current implementation probably checks for 4 packets due to rounding but that shouldn't make a big difference). 

Dynamic power uses `connectionState == disconnected` to boost the power to the max configured level (when armed) and this happens constantly when setting the packet rate to 150Hz and telemetry ratio to 1/8, even at a few meters distance and clear LOS. At other telemetry ratios like 1/64, it is less noticeable because it takes more time to miss 5 packets. 

At 500, 250 and 50Hz, with different telemetry ratios, all is working fine and the connection state doesn't get triggered even through multiple concrete walls.

## Steps to Reproduce

1. Flash the current `master` branch
2. Enable dynamic power
3. Set 150Hz packet rate and 1/8 telemetry ratio
4. Arm (otherwise the `disconnected` state doesn't trigger a boost)
5. Look at the telemetry screen in OpenTX and see that power gets boosted to max a lot. On `master`, the power changes can be quite fast so I set logic condition and special function to play a beep on a power change.

## Possible Solution (Not obligatory)

@AlessandroAU suggested it could be a timing issue with 150Hz. I tried relaxing the timings by changing `SX1280_LORA_CR_LI_4_7` to `SX1280_LORA_CR_LI_4_6` in

https://github.com/ExpressLRS/ExpressLRS/blob/3a18c287b7a5d212be8056397d01f3c7787dfa3b/src/src/common.cpp#L29

but that didn't make a difference for me.

## Details

## Your Environment

- TX hardware: DIY 2.4 E28 and Happymodel ES24TX
- RX hardware: Happymodel EP1 (I tried two)
- Handset model: Jumper T-Lite and Radiomaster TX16S
- OpenTX version (including nightly number): 2.3.15N560
- ExpressLRS version (TX & RX MUST MATCH): current `master` and my branch with dynamic power improvements that doesn't change `UpdateConnectDisconnectStatus()` except for fixing that rounding issue.
- Packet Rate: 150Hz
- Telemetry Ratio: 1/8
- user_defines: 
```
-DRegulatory_Domain_ISM_2400
-DENABLE_TELEMETRY // I just learned this define is now obsolete
-DHYBRID_SWITCHES_8
-DFEATURE_OPENTX_SYNC
-DUART_INVERTED
-DAUTO_WIFI_ON_INTERVAL=60
```

CapnBry, 2021-09-05:
There must be something else going on, I do not think it is because we're not receiving telemetry on 150Hz 1:8. I've got an system running at this speed and modded the TX to spit out what it is doing each timer interval and 100% of the time I see the power boost to max, it immediately following a telemetry packet.
[image]

Probably what you're seeing is due to the telemetry packet being received after the millis is sampled so it looks like the packet is in the distant past so it switches to disconnected because that's definitely a problem I see with this code.

avsaase, 2021-09-05:
Okay yes, that also explains why the TX LQ stays at 100.

closed 2021-09-09 by avsaase