ExpressLRS/ExpressLRS #875: Binding lost with multiple models using Taranis OpenTx

jrwrodgers, 2021-09-05:
## Current Behavior

I have 2 models set up in my Taranis and I have 2 models each with a ELRS receiver. When I switch between models after a flight the new model rx does not bind. 

## Steps to Reproduce

1. Power off model 1

2. Select model 2 on Taranis

3. Power up model 2 - does not bind

## Possible Solution  - Work around!

If I unselect CRSF from the external module on the Taranis, power cycle then select CRSF from external module again then the powered receiver binds. 

- TX hardware: Happymodel TX 2.4GHz
- RX hardware: Happymodel EP2 2.4GHz
- Handset model: Taranis X9D
- OpenTX 2.3.14
- ExpressLRS version (TX & RX MUST MATCH): 1.01
- Packet Rate: 500Hz
- Telemetry Ratio: Off
- user_defines: None

AlessandroAU, 2021-09-14:
This is actually an issue with OpenTX, it doesn't power down the module long enough for a clean reset of the ELRS module, This is fixed in EdgeTX

closed 2021-09-14 by AlessandroAU

simbabreaker, 2021-11-22:
Hi, I have a missing link problem in my ELRS.
This occurs when I move away and when the link breaks, it no longer recovers, when approaching or being next to the Rx.
Only the link is recovered, turning off and reconnecting the Tx and the rx.
My Tx / Rx is a 1st Gen Happymodel in 868.
My FC is a Matek F405 Wing, with Ardupilot.
My Radio a Radiomaster OpenTX, with Yaapu.

What can be the cause of this anomaly.

Thanks