ExpressLRS/ExpressLRS #877: ExpressLRS Team! Please start publish releases!

sandrosov85, 2021-09-06:
Guys, you are gorgeous team!

I really love this project and this incredible Team! You give us lightning speed of making PR and update master branch.  It is insane development speed for opensource project! Many thanks to all of you.

What you think about start using releases instead of merging to master? It would be great practice when i updating (or not) my tx & rx with same version (not master) with known list of features based on release info.

For now some times when i update tx with latest FW and than i update my 3 drones (2 with Frsky r9 & 1 with Happymodel es900rx) i compile fw with version miss match on some rx (yes, you are so fast working but i update fw on rx not so fast because it not always comfortable).

I think time is come for release versions.

With respect to your work. Glad to help project and any of you guys. Sorry for my poor english.

JBKingdon, 2021-09-06:
We already have releases. V1.0.1 is the current stable release. Only use master if you are specifically testing an unreleased feature.

https://github.com/ExpressLRS/ExpressLRS/releases

sandrosov85, 2021-09-06:
Yes and no.

Yes - need stable use 1.0.1

But from version 1.0.1 so many changes and fixes in master so that version non interested for me and other pilots because it have dynamic power, ble joystick and many many other good things.

I think we need some stable version from Master to. May be 1 in 2 weeks would be good practice even for version miss match errors.

JBKingdon, 2021-09-06:
Thanks for your input on release cycles. We'll keep that in mind.

deadbytefpv, 2021-09-06:
Pre-release:
"We need these features NAO. Release it already...."

Release day:
"This is so buggy, why did you release it when you know it's not ready.. "

sandrosov85, 2021-09-06:
Yes, it may be "this it so buggy... " but i can read cosolidated info about release and make decision about update all tx's & rx's.

Many thanks to you team guys.

closed 2021-09-25 by CapnBry