ExpressLRS/ExpressLRS #879: FLIGHT MODE telemetry parameter not present when using Ardupilot

estebanavv, 2021-09-06:
On Ardupilot I get 22 attributes but FLIGHT MODE is missing. Would it be possible to include it?
This was reported before.
https://discuss.ardupilot.org/t/passthrough-telemetry-over-crsf-crossfire/62668/667?u=lanza

schugabe, 2021-09-06:
I don't know why flight mode is missing because this telemetry type is already supported and it looks like ardupilot is using the same frame type as betaflight does.

We current don't send the special ardupilot frame type that is described in #650

StonedDawg, 2021-09-11:
@schugabe i think I can start the test with this for #880

MingchenZhang, 2021-10-03:
I check the ardupilot code and the CSRF there has a [16 byte size](https://github.com/ArduPilot/ardupilot/blob/e1fdab84239e1e2a433f0924bf6954005ee6ebb8/libraries/AP_RCTelemetry/AP_CRSF_Telem.h#L113). The elrs definition is only [10 bytes](https://github.com/ExpressLRS/ExpressLRS/blob/e7cbb6195c462057b865eeea3ffc60ba0cb77fc8/src/lib/CrsfProtocol/crsf_protocol.h#L134). Exls RX just drop that packet since the buffer for it is smaller than the flight mode data in. I tried changing it to 16 and it works. 
create a pull request in https://github.com/ExpressLRS/ExpressLRS/pull/945

CapnBry, 2021-11-04:
This should be resolved by the longer flight mode available in 2.0. I'm going to tag it as closing and close it after 2.0 release unless there is more activity.

closed 2021-12-24 by maybenikhil