ExpressLRS/ExpressLRS #885: I still get a red error when I build any firmwares in China.

imjanuary, 2021-09-08:
## Current Behavior

I still get a red error when I build any firmwares in China. I know we have an especial internet environment. 
Would you please take some time to fix this issue? Thank you!

request to https://raw.githubusercontent.com/ExpressLRS/ExpressLRS/1.1.0/src/user_defines.txt failed, reason: connect ECONNREFUSED 0.0.0.0:443

China

deadbytefpv, 2021-09-08:
Great Firewall

flosean, 2021-09-10:
You need to download the latest branch and build it from local, or you should use VPN to build online.

0crap, 2021-09-20:
@imjanuary Of course nothing the ExpressLRS team can do about.
Ask the great firewall to fix it.
Please close this issue, thank you.

howels, 2021-09-23:
Maybe we can include instructions for offline builds? Perhaps user could download configurator plus a release archive and build from that.

closed 2021-09-26 by imjanuary