ExpressLRS/ExpressLRS #887: [v1.1.0] Lua script shows incorrect build hash

0crap, 2021-09-08:
## Current Behavior

After updating my DIY E28 TX with the latest and greatest v1.1.0, I also grabbed the latest Lua script from the 1.1.x-maintenance branch.
When I run the Lua script all seems to work fine, except the build hash on top of the screen.
It shows 010100 while I expect to see e7cbb61.

## Steps to Reproduce

Flash v1.1.0 on the TX and run the lua script.

## Possible Solution (Not obligatory)

Ignore hash, lua seems to work fine.

## Your Environment

- TX hardware: DIY_2400_TX_ESP32_SX1280_E28_via_UART

0crap, 2021-09-08:
deadbyte tested v1.0.1 with the NEW lua and that shows correct hash.
Problem seems not related to the new lua.

deadbytefpv, 2021-09-08:
I got the same behavior as well.

commit e7cbb61 would show 010100
commit 917bded would show as 010002
1.0.1, even with the new lua from 1.1.0 would show the right hash

It's as if the line https://github.com/ExpressLRS/ExpressLRS/blob/e7cbb6195c462057b865eeea3ffc60ba0cb77fc8/src/lua/ELRS.lua#L383
is not producing the hash.. 
but hey, what do I know? lol

Tested on
- TX hardware: DIY_2400_TX_ESP32_SX1280_E28_via_UART/WiFi
- EdgeTX Nightly (88b8ac86)

JyeSmith, 2021-09-08:
I was changed to version number to help users.  But I guess everyone has been taught to check the hash too well 😄 

Ill update the release notes.

0crap, 2021-09-08:
> I was changed to version number to help users. But I guess everyone has been taught to check the hash too well 😄
> 
> Ill update the release notes.

Well, from a programmers perspective I'm sure this makes totally sense.
But from an end user perspective I can't make anything from it.

Better revert to the old style if you would ask me. :-)

deadbytefpv, 2021-09-09:
> It was changed to a version number to help users. But I guess everyone has been taught to check the hash too well 😄
> 
> Ill update the release notes.

The updates with the new lua implementation has made this part of the code kinda obsolete. Sorry @JyeSmith I know full well how good your intentions were when you wrote the line.. 

It does makes sens to me however: 01.01.00

but yeah, we've been taught very well to read the hash and where to look to check.. :)

0crap, 2021-09-09:
> It does makes sense to me however: 01.01.00
> 
> but yeah, we've been taught very well to read the hash and where to look to check.. :)

Even if you add the dots (which are not shown in the lua screen, I would read the above as v1.01
Not v1.1.0 which I flashed.

Very confusing, but maybe that's only me.
The build hash was fine, no need to change this. ¯\_(ツ)_/¯

SunjunKim, 2021-09-09:
I prefer the hash, which makes zero ambiguity about which variant I'm using.
But I do agree regular users will be happier with the version numbers..
Why don't we have the both?

deadbytefpv, 2021-09-09:
> I prefer the hash, which makes zero ambiguity about which variant I'm using.
> But I do agree regular users will be happier with the version numbers..
> Why don't we have the both?

Should be addressed on next Major release.. so for now, just hang in there.. :)

havoc64, 2021-09-12:
Yep, I too experienced this.  Must have flashed my TX four times over wifi and eventually flashed via USB only to see the 010100,  which to my mind, reflects version  1.0.1.

After the last flash, I powered a quad and found it working and said MEH..

Havoc64

yds, 2021-09-16:
> Why don't we have the both?

yah.. why not both? have the version number and the hash alternate every 5 seconds or so.. both are useful to verify a good flash.. version number is nice, cuz that's a quick confirmation that the new version is what's expected.. the hash is nice for all the reasons everyone else mentioned..

deadbytefpv, 2021-09-17:
As I h

> yah.. why not both? have the version number and the hash alternate every 5 seconds or so.. both are useful to verify a good flash.. version number is nice, cuz that's a quick confirmation that the [...]

As I have mentioned, this should be "fixed" in the next major release.

closed 2021-12-24 by maybenikhil