ExpressLRS/ExpressLRS #892: Find your quad

Moksh-2000, 2021-09-10:
Hello everyone I have an idea since last 2years but I don't give anyone of closed source so elrs is open and now here's my idea when lost the quad and we can't getting gps data or video signal then maybe this feature is worked. like when enable this mode then receiver send a package and inside of packeged a timestamp like 12:1:34:456[x1] and tx received this packaged at 12:1:23:478 [x2] (hour:min:sec:microseconds) now calculate the distance x2-x1= 22microsec now 
m=cs so distance = 6600m = 6.6km this is example and isn't perfect 😅

I know the small error get make huge difference but I'm not one is here , Many people are even smarter than me

Moksh-2000, 2021-09-11:
**here's the solution**

- first radio sent a timestemp like 1:22:33:444 (hour:min:sec:microsec) now number of meter far away receiver received this time stamp and add some delay like 0.6microsec so New time stamp = 1:22:33:444.6 and this signal once again cut number of meter and received remote at 1:22:33:448.6 now we have the time stamp with delay so minus the delay and calculate how much take a time between received time - send time = 2microsec and m=cs 
m= 3×10^8 • 2×10^-6 
m= 6×10^2
m= 600m far away my quad 
[image]

frankjannis, 2021-09-12:
Hi!
The biggest problem I see with your idea is to get the timestamps accurate.
The only way I can Imagine is via GPS. (look [here](https://en.wikipedia.org/wiki/Global_Positioning_System#Timekeeping))
So now your quad must have GPS and has to get the timestamp accurate to the TX. 
(Or your TX needs to have GPS...)

Just sounds like it would be easier to use the already implemented telemetry and 
use the GPS coordinates which are send to your TX directly...

MicroMontis, 2021-09-13:
How do you make sure the TX and RX have the real time clock perfectly synchronized? Also RX does not have RTC...

frankjannis, 2021-09-13:
Hi!
The only way I can Imagine to fix the problem with the RTC is via GPS. (look here https://en.wikipedia.org/wiki/Global_Positioning_System#Timekeeping)
So now your quad must have GPS and has to get the timestamp accurate to the Tx. 
(Or your Tx needs to have GPS...)

Just sounds like it would be easier to use the already implemented telemetry and 
use the GPS coordinates which are send to your Tx directly...

Wouldn't it be a better way of doing this whole thing, by just sending a paket which has to be returned by Rx with as little delay as possible? Tx just has to measure time from sending to recieving (just timing matters and no RTC is needed).
So basicly just like an sonar functions.

Someone would need to check if the link does have low enough jitter and the whole timing could be accurate to - I think - at least 1 µs (~300 meters of signal travel / 2 = 150  meters distace to Rx).

CapnBry, 2021-09-13:
I've mentioned this on Discord but I'll put it here as well, the time doesn't need to be synchronized at all or be a valid wall clock time at all. The TX records its perfcounter, sends the PING command packet and immediately goes to receive, the RX gets a PING and immediately responds with PONG, when the TX gets that it grabs the perfcounter again to calculate round trip time.

The idea is sound, I just think there's too much variability and latency in the packet code for the calculated range to be very precise. If it practically works though, post your code so we can try it out and verify for ourselves! 😀

There's also the option of using the SX1280's Ranging Mode function, which would be incredibly precise but would drop the ELRS connection since it is a special RF mode. That could cause us to disconnect and not be able to reconnect if the LQ was low enough. We're also experimenting with using the SX1281 as a lower-cost replacement for the SX1280, which does not have this function.

Moksh-2000, 2021-09-13:
> Hi!
> The only way I can Imagine to fix the problem with the RTC is via GPS. (look here https://en.wikipedia.org/wiki/Global_Positioning_System#Timekeeping)
> So now your quad must have GPS and has to [...]

Hmm see the @CapnBry is 100% right and he's understand what I'm talking about thank you 🙂
Now @MicroMontis and @frankjannis  we don't need gps because 

My idea is you can used any timestemp or even any packet , you know about how radar work like signal send and touch any surface then get back after we measure how much delay and calculate the distance , same thing is here send any package receiver received after sending same package and we measure how much time delay , (I'm assuming the received and send processing speed = nanosecond so isn't worried about to much)

nangel0, 2021-09-24:
I would like to see this effort towards a proper LoRa ranging mode implementation, also advanced ranging will give you a perfect triangulation of your target.

brandonrc, 2022-01-22:
Added `closing` label because there has been no movement on this feature. 

Please feel free to re-open if you would like to work on this issue. 

I personally like the idea because not all my quads have GPS units but not sure if anyone is interested in getting this to work.

closed 2022-02-06 by github-actions[bot]