ExpressLRS/ExpressLRS #893: New targets not included in 1.1.0

satyagupta, 2021-09-10:
I had submitted a PR for new targets 

https://github.com/ExpressLRS/ExpressLRS/pull/790

It was merged in master, but the new targets are missing from the new release which is 1.1.0

[image]

In configurator select 
Release 1.1.0
Target: QuadKopters 2.4 Ghz
Device: Quadkopters JR 2400 TX

Our device is set to be released by next week. 
By when can we expect the new targets to be available for flashing via configurator. 

Thanks in advance

sohamsp14, 2021-09-12:
Much needed merge in latest release! This will help with more hardware options in the market !

CapnBry, 2021-09-19:
I had heard something about USE_DIVERSITY being explicitly disabled for your hardware, was that something you were going to change?

satyagupta, 2021-09-20:
Yes, the current receiver that we have does not supports diversity but the new receiver would. 
There is sometime for the new receiver to come out, when we plan for that will submit a new PR with necessary changes for the receiver.

CapnBry, 2021-09-20:
Gotcha, just trying to get all the information. There's a good chance we'll be targeting 2.0 before the end of the year and we're probably not going to do two more 1.x.x releases before that. I'm just trying to work out the timeline for what will go where.

satyagupta, 2021-09-20:
Also the PWM receiver that we discussed over DM on discorrd thats on cards as well ;)

CapnBry, 2021-11-01:
Closing as this was resolved in 1.2.0

closed 2021-11-01 by CapnBry