ExpressLRS/ExpressLRS #912: Reciever must be power cycled to connect

skrobo, 2021-09-19:
## Current Behavior

first time I plug drone in it doesn't connect.
unplug the drone and plug it back in it connects instantly

sometimes If I wait about 30 seconds without power cycling it will connect but not always.

## Steps to Reproduce

- I'm not sure but I'm using radiomaster tx16s with the latest edgetx, a brand new happy model tx 2.4, and ep2 reciever
- ELRS version 1.1.0 on rx and tx
- binding phrase= luckyduck (because I have terrible luck when it comes to fpv)

CapnBry, 2021-09-19:
Zero developers can reproduce this at all unfortunately. What packet rate is this? I'd also like to know what the LED on the RX is doing when it doesn't connect. Solid immediately, or slow blinks until it hits wifi mode (super fast blinks)?

Make sure your transmitter is on before plugging in the receiver though, and having it take 12-15 seconds to connect sometimes is unfortunately normal if the first sync is missed by the RX. This is improved in 2.0.

ProfDrYoMan, 2021-09-20:
Same for me with HM EP1 and HM TX.
Trying master also (commit 2cafd7) including #891. Does not help.
Need to unplug and plug again to get a connection after cold boot.
RX led blinks very short once after connect. That's it.
Connecting does take also long since 1.1.0 anyhow. 1.0.1 was way faster.

flosean, 2021-09-21:
Also meet this problem on HM ES24TX and EP RX on 1.1.0 . ( TX16S edgeTX 2.4.0)
It works fine on 1.0.1, every time power up the quad connect quickly.
I have 6 of EP RX and two of them have this problem, not each receiver has this problem.
Need to power cycle again then it can establish connection, but if I keep use the same receiver it doesn't need to power cycle it to connect. Each time I change to another receiver and back to problem receiver again , it need to power cycle to connect.

skrobo, 2021-09-21:
The first time I plug in, my reciever led is usually solid like it’s connected but it is very dim compared to after power cycling and it connecting normally

Aaronsss, 2021-09-22:
Is the led not fully on? If so I had a similar issue when testing binding with the ep2 hooked up to my bench PSU.

Although the ep2 wasn't hitting the 500mA current limit I set I found I had to increase the current limit for it to boot properly 

Additionally I have seen guys in the field getting very frustrated with expresslrs as they were having to keep rebooting their quad

flosean, 2021-09-22:
I solve the problem temporary by trun off the AUTO_WIFI_ON_INTERVAL, and it works fin.

skrobo, 2021-09-22:
This is not an option for me, as I could not get Betaflight passthrough to work and want to be able to update my receiver in the future.

jasonleekennedy, 2021-09-23:
@skrobo - if you don't need full telemetry, telemetry off, wifi on seems to also work for this.

pDR8iLQ4go, 2021-10-03:
I have the same problem using a MATEK F405 CTR and a EP2. First power on doesn‘t work. I always need to reconnect an connect a second time.

grc1000, 2021-10-04:
EP2 LED SOLID after unpacking, I can't update at all. all ways of error. does not appear wifi network. it's just turned on in 5v and gnd.

ProfDrYoMan, 2021-10-04:
Using 1.1.x maintenance does work for me.

grc1000, 2021-10-05:
meu led fica solido quando ligo apenas 5v e gnd, portanto não tem nem como atualizar nenhuma versão. acabei de retirar do pacote. não sei mais o que fazer.

grc1000, 2021-10-05:
my led is solid when I turn on only 5v and gnd, so there's no way to update any version. I just took it out of the package. I do not know what else to do. [ https://drive.google.com/file/d/1Ak8x0gDifIEnMgu_Bz4rlWpDUWzuckDT/view?spm=a2g0s.imconversation.0.0.6c913e5fh2ZwrD&usp=sharing ](url)

pDR8iLQ4go, 2021-10-30:
I've been flashing version 1.2.0 on my EP2. Connection it fine without any delay. No need for 2nd power cycling.

chrisjunlee, 2021-11-02:
Running into this as well. I need to power cycle the RX at least once before it connects to TX.

Strange thing is telemetry connects right away, but control doesn't connect at all.

I've tried waiting several minutes as well, control never establishes.

pDR8iLQ4go, 2021-11-02:
Version 1.2.0?

chrisjunlee, 2021-11-11:
@pDR8iLQ4go yes 1.2.0

closed 2021-11-30 by CapnBry

AlluteTech, 2023-03-15:
I have ExpressLRS 3.2 firmware and I have the same issue. I have to power on the receiver twice to connect to the transmitter. I have this issue after I upgraded the firmware from 2.5 to 3.2. There was no issue on 2.5. My receiver is R9 Slim Plus OTA. I have another model that upgrade to 3.2 but there is no issue. Has anyone seen this issue on 3.2? Note: This is FPV Truck so I do not have FC installed. Instead, I have Crossfire to PWM converter connect to the R9 Slim Plus OTA.