ExpressLRS/ExpressLRS #917: [1.1.0] HappyModel ES900RX boot loop

olegykz, 2021-09-20:
## Current Behavior

Happymodel ES900RX is stuck in a boot loop after flashing `1.1.0`

## Steps to Reproduce

1. Flash `1.1.0` for category: `HappyModel ES900`, device: `HappyModel TX ES900TX`, no matter of flashing method
2. After flashing RX blinks shortly from time to time - no binding, no fallback to wifi, no data at FC, etc.

## Details

Serial log gives the following data in a loop:
```
ets Jan  8 2013,rst cause:4, boot mode:(3,0) 
 
wdt reset 
load 0x4010f000, len 3460, room 16  
tail 4 
chksum 0xcc 
load 0x3fff20b8, len 40, room 4  
tail 4 
chksum 0xc9 
csum 0xc9 
v00057a20 
~ld
```

**Important** - `1.0.1` works fine with the same settings, wiring, etc.

pkendall64, 2021-09-20:
Could you connect to the RX at 420000 baud and see if you can capture the backtrace. If we can get that then we should hopefully be able to get a resolution for the problem.

olegykz, 2021-09-20:
> Could you connect to the RX at 420000 baud and see if you can capture the backtrace. If we can get that then we should hopefully be able to get a resolution for the problem.
> 
> The `rst cause:4` means that the chip has rebooted because of a hardware watchdog timer.

Now I don't get it, it seems that it works fine...
This is the output at 420000
```
ExpressLRS@Module@Booting...
Setting@915MHz@Mode
Setting@915MHz@Mode
Number@of@FHSS@frequencies@=40
p`4`36`2`34`21@23@25@20`35
12@8@1@39@3@28@24`9`13`17
30`22@16@29@5@33@7@26@31`38
19`11`6@37`32@15@10@18`27`14
0@23@37`11@6`26`27@39@14@19
21@25@35`1@17`29@20@33@36`31
2`3`24@15@7@18@16@28@12@4
13`30@10@9`8`22`32@34@5`38
0@16`4@13`33`2`21`20@38@3
35`25`32@28@27@22`29`18@23@12
39@1`14@7@8@24@10@15`5@36
11@31@37@6@30`19@34@26@17`9
p`27`23@34@3@7@6`29@13@18
17@20`5@2@12@14`36@16@9@21
35`30@25@37@26`1`8@33`28@31
24`32@10@4@11`19`15@39@38`22
0@28`1p`5`12`25@37@4@38@33
19@9@20@22@30@6`13@23@11`27
1@16@21@3@14@24`32@2`17@39
26@29@15@36`18@31@35@34`8@7
0@2@23`22@12@17@25`33@21@24
30@19@5@9@20@16@3@6@39`32
38`4@34`14`1`37@7`15@29@31
27@26@10@18@28@36`11@13@35`8
0`12@2@15@13`20@38@37@30@16
19@36@35`4`21@17@
SX127x@Driver@Begin
Hal@Init
12
@found!@(match`by@REG_VERSION@==@0x12)
Setting@ExpressLRS@LoRa@reg@defaults
Setting@ExpressLRS@LoRa`reg`defaults
About`to@start@CRSF`task...
Setting@ExpressLRS`LoRa@reg`defaults
5000
�+�+Setting@ExpressLRS@LoRa@reg@defaults
10ppp
�5�5Setting`Ex�ressLRS@LoRa@reg@defaults
20000
����Setting@Ex�ressLRS@LoRa@reg@defaults
4p00p
����Setting@ExpressLRS@LoRa@reg@defaults
5pp0
�+�+Setting@ExpressLRS@LoRa@reg@defaults
10000
�5�5Setting@ExpressLRS@LoRa`reg@defaults
20000
�l���l��Begin`Webupdater
Stop�ing@Radio
HTTPUpdateServer`ready!`Open@htt�://elrs�rx@in`your`browser
```

I'll flash TX and check the communication

olegykz, 2021-09-20:
Unfortunately (or fortunately?) I'm unable to reproduce the issue - RX is working and communicating fine with TX.
The only thing that I've noticed at `1.1.0` is that wifi way of flashing doesn't work - RX is rebooting as soon as I connect to the wifi hotspot, but, I guess, this is not related to current GH issue.

CapnBry, 2021-09-22:
I have an ES900RX and can't reproduce this. The wifi update also works, but I had the same problem (reboots when a client connects to the wifi) with an EP2 on 1.1.0 so I know that can be a problem, but I can't reproduce that on master.

So can we close this issue if you're not getting a bootloop just from regular operation @olegykz ?

olegykz, 2021-09-22:
Not reproducible, closing.

closed 2021-09-22 by olegykz