ExpressLRS/ExpressLRS #918: Unresponsive EP2 on 1.1.0 with AUTO_WIFI_ON_INTERVAL

trigger7, 2021-09-21:
## Current Behavior

After flashing a Happymodel EP2 receiver with ExpressLRS 1.1.0 and AUTO_WIFI_ON_INTERVAL activated, the RX deoesn't boot properly (solid LED even with TX off) and gets very hot (too hot to touch) in a couple of seconds.

## Steps to Reproduce

1. Flash `1.1.0` for category: `Happymodel 2.4 GHz`, device: `HappyModel EP 2400 RX`, with the defines given below

2. Power up the receiver

3. The LED starts glowing slowly and the RX gets unbearably hot

## Possible Solution (Not obligatory)

Disabling AUTO_WIFI_ON_INTERVAL (same other defines) seems to fix the issue.

## Details

I first tried updating the factory firmware through Wi-Fi (using 10.0.0.1, the configurator is bugged in 1.1.0).
Subsequent attempts were made via Betaflight passthrough (bridging the boot pads) as the RX was unresponsive.

## Your Environment

- TX hardware: Happymodel ES24 Slim Pro
- RX hardware: Happymodel EP2
- Handset model: TBS Tango 2
- OpenTX version (including nightly number): FreedomTX 1.3.0
- ExpressLRS version (TX & RX MUST MATCH): 1.1.0
- Packet Rate: 500 Hz
- Telemetry Ratio: 1:128
- user_defines:
REGULATORY_DOMAIN_ISM_2400
BINDING_PHRASE
LOCK_ON_FIRST_CONNECTION
HYBRID_SWITCHES_8
ENABLE_TELEMETRY
AUTO_WIFI_ON_INTERVAL (= 30)

jasonleekennedy, 2021-09-21:
I am getting the same behavior on an EP1 at 250Hz, if I connect power, disconnect, reconnect it starts up fine even with auto_wifi_on (same user defines).  I also tested a different UART to make sure it wasn't SBUS related.

JBKingdon, 2021-09-21:
Looks like a dup of https://github.com/ExpressLRS/ExpressLRS/issues/912
Could you check that 912 describes the problem you're seeing and close this one if so?

Aaronsss, 2021-09-22:
Is the led not fully on? If so I had a similar issue when testing binding with the ep2 hooked up to my bench PSU.

Although the ep2 wasn't hitting the 500mA current limit I set I found I had to increase the current limit for it to boot properly

trigger7, 2021-09-22:
I didn't try to power cycle the receiver while I was having this issue, so I can't confirm my issue is related to #912 

My main concern is the temperature the receiver is reaching, and how quickly it's heating up.
It may be a hardware design flaw (having a wifi chip in such a small pcb), in which case I wouldn't be confortable enabling wifi at all on EP-series RXs.
If it's a software issue, I think it needs to get looked at.

closed 2021-11-30 by CapnBry