ExpressLRS/ExpressLRS #921: EP2 doesn't want to work with SucceX E F405 Fc

Marcelogix, 2021-09-22:
## Current Behavior

At first I tried to connect the EP2 with the Fc of the Nazgul 5 v1 as with my Diatone L4.
With the Diatone L4 it works perfectly fine. On the SucceX E F405 (Nazgul FC) it got stock in Bootloader. 
After searching for a while I found #655 and tried to use a pull-up resistor but it didn't work for me. 
When I power cylce the Rx while the FC is still connected it works perfectly fine (using binding with radio) until I unplug the whole FC.
As i didnt want to power cycle the Rx everytime with the FC still powered up, I moved to another Uart. On Uart6 it works sometimes after power cycling as in #912. Sometimes i need to connect the battery 2-3 times, sometimes way more often.
Everytime the Rx doesn't work, the led is dimmed (I used this as an Indicator).
I recognised that the EP2 gets quiet hot sometimes (as in #918). I didn't figure out wether its just normal or when wifi turned on.
After several times (everytime cooled back down to normal temperature) the receiver instantly broke and wouldn't turn back on.
I tried this twice and both EP2's burned out (first after 4-5 tries and second after ~25 tried).

Just ordered some more for testing

## Steps to Reproduce

1. Connect EP2 to Uart6 instead of Uart2

2. Connect the battery/usb cable

3. Power Cycle until the dimmed LED gets bright

4. After a lot of testing it burns ;)

## Possible Solution (Not obligatory)

At first receiver the ESP8285 broke. Maybe disabling wifi fixes the problem if its an overheating issue.

## Details

Tried with version 1.0.1 and 1.1.0 but left auto wifi always on (using 40 secs, so it shouldnt turn on usually?)

## Your Environment

- TX hardware: Happymodel ES24TX
- RX hardware: HappyModel EP2 2.4 GHz
- Handset model:  RadioMaster Tx16s and RadioKing Tx16s (using T18 software)
- OpenTX version (including nightly number): 2.3.14-otx
- ExpressLRS version (TX & RX MUST MATCH): 1.0.1 and 1.1.0
- Packet Rate: 250Hz
- Telemetry Ratio: 1:64
- user_defines:
        REGULATORY_DOMAIN_ISM_2400
        BINDING_PHRASE
        LOCK_ON_FIRST_CONNECTION
        HYBRID_SWITCHES_8
        ENABLE_TELEMETRY
        AUTO_WIFI_ON_INTERVAL (40s)

Marcelogix, 2021-09-22:
[image]

This is the first Ep2 which broke. The second one doesnt have these obvious spots

Magginator98, 2021-09-22:
The second Ep2: The SDG gets hot here...
[image]
[image]
.

djgoldeneye, 2021-09-27:
i had this same issue. replaced by rotorvilllage. the replacement works better but still every so often i have a "sensor lost" message. im also having an issue where my quad falls out of the sky. Using Tx16 radio. Tried using my radiomaster R81 Rx didnt have an issue with that Rx. Except for very poor range and penetration.

Whenever the quad dies, theres a voltage drop, Could this be related.?

closed 2021-11-30 by CapnBry