ExpressLRS/ExpressLRS #925: Add SBUS protocol

kimman1, 2021-09-23:
I think we should add SBUS output protocol for some reason for using it easy. Because some Tiny FC don't have both TX and RX just only have RX 

Is a good idea for enhancement ? Or just is a bad idea 

Thank for reading ^_^

deadbytefpv, 2021-09-23:
There's an incoming feature that inverts the receiver's TX pad so that you can connect an ExpressLRS receiver's TX pad into an inverted RX pad in a UART-Limited FC (e.g. SBUS pad).

This isn't SBUS protocol though, but still CRSF Protocol..

kimman1, 2021-09-24:
> There's an incoming feature that inverts the receiver's TX pad so that you can connect an ESP-based ExpressLRS receiver's TX pad into an inverted RX pad in a UART-Limited FC (e.g. SBUS pad).
> 
> This isn't SBUS protocol though, but still CRSF Protocol..
> 
> See: #878
> 
> Edit: Added the relevant PR. 
Many thanks, Deadbyte

I've another idea and I think if can be possible ? 
Okay, we have ESP8266/8285 on RX side so we will have wifi, my idea to using wifi to passthrough UART so we can using for configurating on speedybee

CapnBry, 2021-09-25:
> Okay, we have ESP8266/8285 on RX side so we will have wifi, my idea to using wifi to passthrough UART so we can using for configurating on speedybee

This needs support in Betaflight. CRSF protocol doesn't allow the size of packets needed for MSP over Telemetry to hold the Betaflight Configurator / iNav Configurator / SpeedyBee data. I've spent a lot of time writing the ELRS side but it needs every FC software to be updated.

kimman1, 2021-09-25:
> > Okay, we have ESP8266/8285 on RX side so we will have wifi, my idea to using wifi to passthrough UART so we can using for configurating on speedybee
> 
> This needs support in Betaflight. CRSF protocol doesn't allow the size of packets needed for MSP over Telemetry to hold the Betaflight Configurator / iNav Configurator / SpeedyBee data. I've spent a lot of time writing the ELRS side but it needs every FC software to be updated.
> 
> If this Issue is about just using an SBUS pad on a FC and not actually SBUS protocol (as it seems), then that is done already and you can close this issue.

Thank you very much @CapnBry

closed 2021-09-25 by kimman1