ExpressLRS/ExpressLRS #938: Port RX to $10 ready to use E78-868LN22S (ASR6501 SX1262 modules)

krzysztofkuczek, 2021-09-29:
Hi,
What do you think about using such a ready to use modules instead of 400% overpriced stuff?
I found out how to program and debug, you can find more details here
https://qczek.beyondrc.com/e78-868ln22s-asr6501-module-swd-programming/

I think that it has enough resources to run Express LRS :)
Anyway if someone from ELRS team will not do it, I will add such a "Espress LRS compatibility" feature to my Qczek LRS ;)

*Instead of commenting "I want this too", consider leaving a :+1: instead, to reduce spam and clutter*

0crap, 2021-10-02:
The module E78-868LN22S spec is 22dBm, which means on ExpressLRS it will only output around 100mW without blowing up.
The module only supports 900MHz and we have already plenty of cheap DIY and off the shelf products for this range.
For receivers (you use many) it matters if a part is a bit better priced, for a TX (you use only one) not so much.
Porting means a lot of effort and extra code to maintain.

All in all it does not seem something worth the effort.
Just my two cents.

brandonrc, 2022-01-07:
I am just going to close this issue because most of the developers do not run 900mhz and I think it is a slim chance they will tackle this. 

If someone want's to do the work to get this part working with ELRS please open this back up and attach the Pull Request.

closed 2022-01-07 by brandonrc