ExpressLRS/ExpressLRS #948: ES900TX binds only few times after flash + LQ drops

sohamsp14, 2021-10-03:
I have been noticing a weird problem with my ES900TX. After I flash the receiver, it binds to the TX only 2-3 times and then it stops binding. I have to reflash again get it to bind again. 
Also, there are continuous LQ drops with the radio being directly beside the RX as well. This is seen in the goggles as RXLOSS accompanied by micro failsafes with an eventual complete failsafe. (As happened in today's flight)
The RX was working fine on the previous FW. I recently updated to 1.1.0 and this issue has been popping up. But my other receiver which is also on 1.1.0 seems to be working fine. 

Video of problem:
[Youtube Link](https://youtu.be/gjva3aevqq8)

RX Settings
[image]

TX Settings
[image]

deadbytefpv, 2021-10-04:
Can you flash the RX with the branch 1.1.x-maintenance, please?

sohamsp14, 2021-10-04:
Did not work. Put in another receiver eventually and the LQ seems to be much better. >=97 at close range (although this should be 100 ideally). So probably a bad receiver? Will do a field test soon.

sohamsp14, 2021-11-01:
Changing the receiver fixed the issue. My guess is it was a hardware problem. Closing

closed 2021-11-01 by sohamsp14