ExpressLRS/ExpressLRS #1062: Support for Flysky Paladin PL18 module FRM301 (SX1281).

porfiheli, 2021-11-06:
Hello everyone. Thanks for the great work with ELRS.

I have bought for a couple of weeks a [Flysky Paladin PL18](https://www.flysky-cn.com/pl181englishzj). The handset is great. It is really likely that it will get in the near future EdgeTx. I would like to use it with the ExpressLRS receivers. I disassembled the external module ([FRM301](https://www.flysky-cn.com/frm301-canshu)) and tried to find out the schematics. I have the [Flysky Noble NB4](https://www.flysky-cn.com/nb4) too and as far as i can say, the [RF module](https://fccid.io/N4ZFG400/Internal-Photos/Internal-photos-4221605) (internal) has the same layout. The AFHDS3 receivers have the same RF module but the layout must be different.

- The RF Module (FS-HF200)  is based on the Semtech SX1281. [Pictures of the module](https://fccid.io/N4ZFTR1000/Internal-Photos/Internal-Photos-4057548). 
- The uC is the STM32F030C8T6.

Would it be possible to port the code from another stm32 tx like the FM30 or Ghost? I would like to help coding but i don't have much experience and i don't know if the uC has enough memory.

I suppose that for support this new target a new variant would be needed in `/src/variants/` and in boards a new one should be added, right? If someone could help me to get this module supported would be really appreciated. 

[image]

---
Here the schematics. If something is missing i can try to trace the pinout.

[2021-11-06 FRM301 Schematics.pdf](https://github.com/ExpressLRS/ExpressLRS/files/7491100/2021-11-06.FRM301.Schematics.pdf)
*Instead of commenting "I want this too", consider leaving a :+1: instead, to reduce spam and clutter*

porfiheli, 2021-11-09:
RCC configuration.
[image]

porfiheli, 2021-11-16:
Sport Connection is like that:

[image]

I suppose that i have to set UART1 as half-duplex and inverted, right? @CapnBry you did the port for the FM30 TX. Could you please help me with this module? I get no crfs communication between the module and the handset.

Maybe the 8kB ram of the f030 are not enough for elrs tx code? i had to change the min stack size from 0x1000 to 0x400 to fit in the ram. (_Min_Stack_Size = 0x400)

CapnBry, 2021-11-16:
Does the F030 have builtin hardware inversion? Does the handset do inversion? In the FM30 it is an F103 so all the muxing of the data and inversion is handled with external circuitry and we use GPIOs to toggle the inverter and direction. You're probably going to need to get a scope on it and figure out what side isn't doing what it is supposed to. I don't think I'd be able to help without the hardware though.

porfiheli, 2021-11-16:
Thanks  @CapnBry for the fast reply. 

In the [reference manual](https://www.st.com/resource/en/reference_manual/dm00091010-stm32f030x4x6x8xc-and-stm32f070x6xb-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) page `628` is the uart control register. If it is in the st documentation, the hardware inversion should be available. Externally there is no other hardware, the sport is connected through 2 in series resistor to rx and tx pin of uart1. I will try to check it with the scope to see what is happening. 

[image]
[image]

```
/******************  Bit definition for USART_CR2 register  *******************/
#define  USART_CR2_ADDM7                     ((uint32_t)0x00000010)            /*!< 7-bit or 4-bit Address Detection */
#define  USART_CR2_LBDL                      ((uint32_t)0x00000020)            /*!< LIN Break Detection Length */
#define  USART_CR2_LBDIE                     ((uint32_t)0x00000040)            /*!< LIN Break Detection Interrupt Enable */
#define  USART_CR2_LBCL                      ((uint32_t)0x00000100)            /*!< Last Bit Clock pulse */
#define  USART_CR2_CPHA                      ((uint32_t)0x00000200)            /*!< Clock Phase */
#define  USART_CR2_CPOL                      ((uint32_t)0x00000400)            /*!< Clock Polarity */
#define  USART_CR2_CLKEN                     ((uint32_t)0x00000800)            /*!< Clock Enable */
#define  USART_CR2_STOP                      ((uint32_t)0x00003000)            /*!< STOP[1:0] bits (STOP bits) */
#define  USART_CR2_STOP_0                    ((uint32_t)0x00001000)            /*!< Bit 0 */
#define  USART_CR2_STOP_1                    ((uint32_t)0x00002000)            /*!< Bit 1 */
#define  USART_CR2_LINEN                     ((uint32_t)0x00004000)            /*!< LIN mode enable */
#define  USART_CR2_SWAP                      ((uint32_t)0x00008000)            /*!< SWAP TX/RX pins */
#define  USART_CR2_RXINV                     ((uint32_t)0x00010000)            /*!< RX pin active level inversion */
#define  USART_CR2_TXINV                     ((uint32_t)0x00020000)            /*!< TX pin active level inversion */
#define  USART_CR2_DATAINV                   ((uint32_t)0x00040000)            /*!< Binary data inversion */
#define  USART_CR2_MSBFIRST                  ((uint32_t)0x00080000)            /*!< Most Significant Bit First */
#define  USART_CR2_ABREN                     ((uint32_t)0x00100000)            /*!< Auto Baud-Rate Enable*/
#define  USART_CR2_ABRMODE                   ((uint32_t)0x00600000)            /*!< ABRMOD[1:0] bits (Auto Baud-Rate Mode) */
#define  USART_CR2_ABRMODE_0                 ((uint32_t)0x00200000)            /*!< Bit 0 */
#define  USART_CR2_ABRMODE_1                 ((uint32_t)0x00400000)            /*!< Bit 1 */
#define  USART_CR2_RTOEN                     ((uint32_t)0x00800000)            /*!< Receiver Time-Out enable */
#define  USART_CR2_ADD                       ((uint32_t)0xFF000000)            /*!< Address of the USART node */
```

CapnBry, 2021-11-16:
Should be doable then. We use those flags on F3 to do inversion in software so should be the same code. I just wasn't sure if the F0 had it or not.

porfiheli, 2021-11-25:
I get communication with the module but the lua script take sometimes a lot of time to load completely. 

[image]

SUPANOOB-ZA, 2022-02-23:
Hi there, 
Just to clarify, did you manage to get ELRS to work using the TX ?

gagarinlg, 2022-09-02:
I would be very interested in getting a firmware for the FRM301 that does ELRS. My PL18 runs on EdgeTX.

gagarinlg, 2022-09-02:
I got the old PR updated and now two people have PL18 running EdgeTX with working ExpressLRS on the FRM301

porfiheli, 2022-09-02:
Hi @gargarinlg, nice to read that it is working now. I had unfortunately no time to finish the project. I will try to update my PL18 to Edge-TX and then will update the FRM301, will be great to have ExpressLRS 3 with the PL18 :-)

gagarinlg, 2022-09-02:
It is all experimental right now! 
There is a manual available to bring EdgeTX on the PL18:
https://github.com/EdgeTX/edgetx/wiki/Flashing-EdgeTX-to-Flysky-PL18-or-Paladin-EV

SUPANOOB-ZA, 2022-11-13:
@gagarinlg please help me flash ELRS to the flysky FRM302 .... what do I need to do?

gagarinlg, 2022-11-26:
> @gagarinlg please help me flash ELRS to the flysky FRM302 .... what do I need to do?

It does not work since I rebased to the 3.x version of the code.

zandorsp, 2022-12-16:
> Hello everyone. Thanks for the great work with ELRS.
> 
> I have bought for a couple of weeks a [Flysky Paladin PL18](https://www.flysky-cn.com/pl181englishzj). The handset is great. It is really likely that it will get in the near future EdgeTx. I would like to use it with the ExpressLRS receivers. I disassembled the external module ([FRM301](https://www.flysky-cn.com/frm301-canshu)) and tried to find out the schematics. I have the [Flysky Noble NB4](https://www.flysky-cn.com/nb4) too and as far as i can say, the [RF module](https://fccid.io/N4ZFG400/Internal-Photos/Internal-photos-4221605) (internal) has the same layout. The AFHDS3 receivers have the same RF module but the layout must be different.
> 
>     * The RF Module (FS-HF200)  is based on the Semtech SX1281. [Pictures of the module](https://fccid.io/N4ZFTR1000/Internal-Photos/Internal-Photos-4057548).
> 
>     * The uC is the STM32F030C8T6.
> 
>     * HSE 12 MHz.
> 
> 
> Would it be possible to port the code from another stm32 tx like the FM30 or Ghost? I would like to help coding but i don't have much experience and i don't know if the uC has enough memory.
> 
> I suppose that for support this new target a new variant would be needed in `/src/variants/` and in boards a new one should be added, right? If someone could help me to get this module supported would be really appreciated.
> 
> [image]
> 
> Here the schematics. If something is missing i can try to trace the pinout.
> 
> [2021-11-06 FRM301 Schematics.pdf](https://github.com/ExpressLRS/ExpressLRS/files/7491100/2021-11-06.FRM301.Schematics.pdf) _Instead of commenting "I want this too", consider leaving a +1 instead, to reduce spam and clutter_

The new Flysky EL18 [https://www.flysky-cn.com/el18-specifications](url) will have a internal AFDHS3 too and may work.

gagarinlg, 2022-12-16:
It should be possible, but RAM might be an issue.
I have started porting the code to the Flysky EL18 RF part.

PlanetSnurfer, 2022-12-26:
Pls let us know, how the porting to the internal Modul of EL18 is going. I would try it immediately.

PlanetSnurfer, 2023-04-07:
> It should be possible, but RAM might be an issue.
> I have started porting the code to the Flysky EL18 RF part.

Do you have any news?

ishampadron, 2024-01-29:
I have an EL18 and I would be open to testing if there is a path to get ELRS in the internal module.

richardclli, 2024-06-13:
New development about ELRS for Flysky TX modules will be followed here:

https://github.com/ExpressLRS/ExpressLRS/issues/2778

closed 2024-07-17 by JyeSmith