ExpressLRS/ExpressLRS #1066: Lua cmd struct member missing

xtr0d3m0n, 2021-11-07:
## Current Behavior

Error while compiling Lua files

## Steps to Reproduce

Compile Master Branch with VERBOS BEBUG enabled

this log macro tries to access a missing struct member on cmd
DBGVLN("sending response for [%s] chunk=%u step=%u", cmd->name, nextStatusChunk, cmd->step);

In file included from lib/LUA/lua.cpp:5:0:
lib/LUA/lua.cpp: In function 'void pushResponseChunk(luaItem_command*)':
lib/LUA/lua.cpp:155:61: error: 'struct luaItem_command' has no member named 'name'
   DBGVLN("sending response for [%s] chunk=%u step=%u", cmd->name, nextStatusChunk, cmd->step);
                                                             ^
lib/logging/logging.h:47:26: note: in definition of macro 'DBGLN'
       debugPrintf(msg, ##__VA_ARGS__); \
                          ^
lib/LUA/lua.cpp:155:3: note: in expansion of macro 'DBGVLN'
   DBGVLN("sending response for [%s] chunk=%u step=%u", cmd->name, nextStatusChunk, cmd->step);
   ^
*** [.pio\build\EXPRESS_TX\lib461\LUA\lua.cpp.o] Error 1
1.

2.

3.

4.

## Possible Solution (Not obligatory)

remove the struct access, Not sure if the cmd ->name is needed, Was this struct member got removed by a mistake ?

## Details

## Your Environment

Windows

- TX hardware:
- RX hardware:
- Handset model:
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH):
- Packet Rate:
- Telemetry Ratio:
- user_defines:

xtr0d3m0n, 2021-11-07:
It seems like there is no mention of name anywhere else. Seems like it was left there by mistake. Should I just remove it at commit a fix to master?

pkendall64, 2021-11-07:
Yes it's definitely leftover from some earlier refactoring work

xtr0d3m0n, 2021-11-08:
https://github.com/ExpressLRS/ExpressLRS/pull/1070
Fixed on this pr can somebody review?

commit 15f2ac3d (2021-11-08): Fixed #1066 (#1070)
closed 2021-11-08 by JyeSmith