ExpressLRS/ExpressLRS #1068: Namimno STM32 not saving EEPROM settings between flashes

wvarty, 2021-11-07:
## Current Behavior

After a FW flash, the settings in LUA are back to defaults. Prior to the flash, I had rate = 250hz, and after it was default 500hz.

## Steps to Reproduce

1. Flash master to Namimno ST32

2. In LUA, set 250hz, or something other than 500hz.

3. Flash master to Namimno ST32

4. Verify that rate is now 500hz in LUA

## Details

Is this behaving differently on STM32 vs ESP32? I suspect the EEPROM lib isn't handling the erase properly.

## Your Environment

- TX hardware: Namimno flash STM32 version
- RX hardware: NA
- Handset model: T16
- OpenTX version (including nightly number) 2.3.14
- ExpressLRS version (TX & RX MUST MATCH): Master branch

CapnBry, 2021-11-09:
Fixed by #1071?

brandonrc, 2021-12-20:
@maybenikhil verified, fixed.

closed 2021-12-20 by brandonrc