ExpressLRS/ExpressLRS #1075: FalcoX FW2823

keisukesamba, 2021-11-09:
I'm using the Revolt OSD (FW2823).
I can bind, but the switch and stick are not recognized by this firmware.

When I use the latest firmware, it works, but I want to use it with FalcoX2823 FW.
I've tried changing the UART, downgrading the ELRS version, etc., but no luck.

Do you have any suggestions?

FC: Revolt OSD or Revolt V3
FC firmware: 2823
TX: Happymodel 2.4 V2board
RX: EP2
taranis X9DPlus SE 2019
OpenTX 2.3.14

ExpressLRS Configurator v1.2.3
TX Version 1.2.0
RX Version 1.2.0

schugabe, 2021-11-10:
Can you try to enable the option -DUART_RX_SPEED_400K in this line of the user defines: https://github.com/ExpressLRS/ExpressLRS/blob/1.2.x-maintenance/src/user_defines.txt#L49

keisukesamba, 2021-11-10:
thanks ！
I tried it, but it didn't work. What if I lower the baud rate further?

Will there be a difference between odd and even ports?

schugabe, 2021-11-10:
I should have put more info in the first comment: you need to compile and upload the RX with this option or this option https://github.com/ExpressLRS/ExpressLRS/blob/master/src/user_defines.txt#L47 if you are trying the RC1 of 2.0 the option does not change anything on the TX module.

Otherwise you simply have to follow the regular "wire + setup a crossfire rx" on your FC. I don't have a revolt FC so I can't try it myself. Lowering the speed is necessary on a KISS v1 so I wanted to try if the lower uart speed works in your case. Lowering even further does not help.

keisukesamba, 2021-11-10:
thank you!

I rewrote RX after seeing the previous post!
If I'm using 1.2.0, do I need to rewrite TX as well?

schugabe, 2021-11-10:
No that settings only changes the UART config on the RX

keisukesamba, 2021-11-10:
thank you! try it!

brandonrc, 2022-01-24:
Closing this issue because the OP has not added any more information

closed 2022-01-24 by brandonrc

yyu-00, 2024-12-28:
Who can share this few with me ? Thank  you