ExpressLRS/ExpressLRS #1076: Selecting power level can trigger radio emergency mode (2.0.0-RC)

Isopodus, 2021-11-09:
## Current Behavior
Selecting the power level via elrsV2.lua script causes radio emergency mode if clicking the "BACK" menu item **while the progress bar is not finished animating.**

## Steps to Reproduce
1. Get the gear (list below)
2. Update to 2.0.0-RC
3. Run elrsV2.lua script
4. Press **> TX Power**
5. Press and select any **Max Power** level (including the one you're currently on e.g. 10 mW)
6. Quickly press the **[----BACK----]** button (fast enough so the progress bar did not finish yet)
7. See the emergency mode being activated

## Possible Solution
Disable the BACK button until progress bar finishes? Idk if possible...

## Details
(Tested on the desk)
As the emergency mode triggers I can hear the TX module fan stops and turns on back again, possibly indicating module reboot. The armed quad looses the signal and catches a failsafe. The TX-RX link then get's active again (RX led solid red, quad controls are back normal).

The quad will get a failsafe anyway, so not suitable for systems without a GPS rescue (however who will ever fly a long range without a GPS?). Anyway, such case is possible so I think this needs to be fixed before the actual release...

Thanks!

## Your Environment

- TX hardware: Radiomaster TX16S + Happymodel ES900TX
- RX hardware: Happymodel ES900RX
- OpenTX version: opentx-tx16s 2.3.14-otx (1e09791a)
- ExpressLRS version: 2.0.0-RC
- Packet Rate: 200 Hz
- Telemetry Ratio: 1:64

CapnBry, 2021-11-09:
I can confirm this:
* ES900TX will panic if the --BACK -- item is pressed when the list is reloading. Workaround: Use the physical Back/Exit/RTN button on the handset to go up the menus
* Using the --BACK-- item on any platform causes the loading to pause. Workaround: Use the physical back button.

I'll work on both of these.

CapnBry, 2021-11-12:
Should be fixed in RC2, can you test and close if it does?

Isopodus, 2021-11-13:
Haha, yes, can't trigger that anymore, thanks for a quick fix!

PR #1080 (2021-11-11): Prevent crash when using BACK item when lua is reloading
Fixes #1076

### Details
The root cause here was that the code in Lua was sort of a mishmash where it would set up to just request an update to the single field that was changed, then refresh all the fields if it was an editable field. Since the back button is not editable it would just request the `backbuttonId` field over and over and over again until the TX would respond or another field is edited. Of course, that field is not valid on the TX so the TX would crash or send garbage data.

### Fix
I'm not sure why the Lua reloads all fie [... cut]
closed 2021-11-13 by Isopodus