ExpressLRS/ExpressLRS #1082: BLE Joystick mode not working (2.0.0-RC1)

thkn-hofa, 2021-11-09:
## Current Behavior

Cannot use the BLE Joystick functionality

## Steps to Reproduce

1. Select model that uses the ExpressLRS module (CRSF)
2. Open LUA script and select BLE Joystick
3. Press ENTER when 'Start BLE Joystick?' pops up
4. 'Joystick Running...' pops up and disappears after three seconds
5. Cannot find the TX module on multiple devices

## Possible Solution (Not obligatory)

Magic?

## Details

CapnBry mentioned that the ELRS radio signal is turned off when going into Bluetooth but the LUA script still reads 0/150 in the top right corner and the module gets warm after a few minutes which leads me to believe it's not switching to bluetooth.

## Your Environment

- **TX Hardware**: BetaFPV 2.4 Nano RF (first generation)
- **RX hardware**: Windows 10 (Intel Bluetooth) & OnePlus 8T 
- **Handset model**: FrSky Taranis X9-lite
- **OpenTX version (including nightly number)**: 2.3.14-otx (1e09791a)
- **ExpressLRS version (TX & RX MUST MATCH)**: 2.0.0-RC1 (6f798a)
- **Packet Rate**: 150Hz
- **Telemetry Ratio**: 1:32
- **user_defines**: /

thkn-hofa, 2021-11-09:
Update: after starting BLE Joystick, the top right briefly reads 0/0, then 0/300 and then back to 0/150. Is something being reset?

pkendall64, 2021-11-10:
The spinner should stay spinning. If you press the return button it will cancel the joystick. If you long press it will exit LUA.
The joystick internally changes the packet rate to 250Hz, so if you exit LUA then go back in you should see it at 0/250.
The reason you see the numbers bounce is because it has restarted the module because you have exited BLE mode.

thkn-hofa, 2021-11-10:
Thanks for the extra info. I don't manually exit the BLE Joystick mode or LUA script and it never goes to 250Hz. The 'Joystick Running...' automatically disappears after 3 seconds.

pkendall64, 2021-11-11:
Can you take this to discord help-and-support channel

thkn-hofa, 2021-11-11:
@wvarty on the Discord channel tested it out and he can't use it, too. WiFi on that TX module doesn't work and the ESP32 uses the same functionality for WiFi and BLE so chances are great that it's BetaFPV bad implementation to blame.
Does anyone know how to contact BetaFPV directly? 😅

sherghan, 2021-11-13:
Same problem here, on 2.0.0-rc2.
BetaFPV Nano TX Module.

dupeiran001, 2021-11-23:
betafpv nano v1.0.0, elrs v2 rc2, edge tx 2.4
same problem occur

nightflasher, 2021-11-26:
Found this library, for esp32-ble-gamepad
https://github.com/lemmingDev/ESP32-BLE-Gamepad
Maybe this can be used instead of the default ble lib?

AlessandroAU, 2021-11-26:
We are using the same BLE lib as that already

BRM-Code, 2022-01-05:
I have this problem as well.
Frsky Xlite, EdgeTx 2.5, BetaFpv Nano Tx, ELRS 2.0.1
Is it BetaFpv's fault this isn't working?

thkn-hofa, 2022-01-05:
Yes, I'm wondering if this ticket should be closed or stay open so as to not create more tickets for this.

brandonrc, 2022-01-07:
Regular is under spec and the module browns out.

Please contact BetaFPV or the company that you purchased the product from.

closed 2022-01-07 by brandonrc