ExpressLRS/ExpressLRS #1087: LUA V2 throwing 'Script Panic' message and not working

BludzUK, 2021-11-10:
## Current Behavior

When I try to run the LUAV2 Script it says Script Panic and doesn't run

## Steps to Reproduce

Run the LUA V2

## Possible Solution (Not obligatory)

Incompatability issue with X9D+

## Details

Managed to flash with release candidate using V1 Lua, but V2 doesn't work so can't see the data :(

## Your Environment
I'm running EdgeTX 2.5 on a Taranis X9D+
HiYounger TX module (BetaFPV target/clone)

pkendall64, 2021-11-10:
Can you add a picture of the screen with that. Are you sure you downloaded the script correctly? Github UI is a bit crap about downloading raw files and catches a few people out. Grab it from this URL https://raw.githubusercontent.com/ExpressLRS/ExpressLRS/master/src/lua/elrsV2.lua

BludzUK, 2021-11-10:
[image]

BludzUK, 2021-11-10:
99% sure I got the script downloaded fine, I saw it catches some people out. I right clicked and saved target as, produced elrsV2.lua file.

BludzUK, 2021-11-10:
I just put TBS AgentLite on my radio and it is working. Really cool that you got it to work so well with it, well done you cheeky buggers hahahahaha. 
Hopefully that can help you fix this bug.

BludzUK, 2021-11-10:
Okay.... so now that I have the TBSAGENTLITE folder (that comes in the zip file from TBS) in the SCRIPTS/TOOLS/ directory, the ExpressLRS LUA V2 script is now working.
So it must be relying on one of the files contained in that folder and you either need to copy it and bundle with the LUA V2, or tell people to have AgentLite on the SD card.
Hope I helped in some small way :)

BludzUK, 2021-11-10:
Okay cool, well at least it's sorted now. Hope I didn't cause offense by presuming it depended on the Agent Lite folder, you can see how I would draw that conclusion. Looking forward to testing this out at the weekend :)
Thanks for the fast response :)

pkendall64, 2021-11-10:
No offence, just wanted to make it clear that in no way would we create a dependency on a commercial piece of software.

closed 2021-11-11 by pkendall64