ExpressLRS/ExpressLRS #1088: RX lost when LUA script activated on X-lite running at 115k baud (2.0.0-RC1)

dopplet, 2021-11-11:
## Current Behavior

X-lite radio with TX Slim (Ironman), EP2 RX.  Flashed to 2.0.0-RC1.  RX connects fine to TX, all works fine until LUA (V2 script from GITHUB) is started, RX immediately disconnects, and never reconnects when LUA is exited.  Requires rx restart.  RX start when LUA already active, connect and instant disconnect.

X-lite was at default 115k baud

At 400k baud, this issue does not happen.  

This radio combination works perfectly fine on 1.2.0

## Steps to Reproduce

Start TX
Start RX
Activate LUA
## Possible Solution (Not obligatory)

400k baud is a workaround, however it should work with 115k as 1.2.0 did.

## Details

No additional details

## Your Environment

- TX hardware: Happy Model 2400 TX Slim (IRONMAN)
- RX hardware: EP2 RX
- Handset model: Frsky X-Lite 
- OpenTX version: (Edge TX 2.50)
- ExpressLRS version (TX & RX MUST MATCH): 2.0.0-RC1
- Packet Rate: 250
- Telemetry Ratio: 1:8
- user_defines:

dopplet, 2021-11-11:
Additional info.  The X-lite was recently upgraded to 2.5.0, it's not my daily driver so hasn't been used.  Someone on the discord asked me to try 115k at 50hz, this worked.  I tried 150, works, and then 250, works.  All works now.  I think the radio was in an unknown baud rate from the edgetx upgrade, and actively setting it put it to a proper known state....  likely nothing wrong with elrs.

pkendall64, 2021-11-11:
That was me! 😆

dopplet, 2021-11-11:
Closing issue, I am not worthy!!!!!

closed 2021-11-11 by dopplet