ExpressLRS/ExpressLRS #1089: DISABLE_STARTUP_BEEP doesnt work on v2rc1

Directory, 2021-11-11:
template not needed. r9m. 2 annoying beeps on crsf connection establishment.

pkendall64, 2021-11-11:
This is working as intended. Theres just no option to disable all beeps. If you'd like that to be an option then perhaps adjust this ticket and label it as a feature request.

Directory, 2021-11-11:
I repeat, this did not happen on v1.0.0

pkendall64, 2021-11-11:
> I repeat, this did not happen on v1.0.0

That was a bug in 1.0

Directory, 2021-11-11:
then how come the 2 beeps do not play when disable_startup_beep is disabled and i hear that obnoxious melody. they only play when the startup beep is disabled.

pkendall64, 2021-11-11:
They do play, they are the extra 2 beeps at the end of the startup melody.

Directory, 2021-11-11:
they sound different. with default melody enabled its a low tone and then a high tone. but with it disabled its a high tone followed by a low tone. theres no reason for the crsf connection establishment tone to change.

pkendall64, 2021-11-11:
What it does is play the melody, if enabled, and we see a state change from starting to disconnected so we queue a disconnected tune, then we get connected so we queue the connected tune (overwriting the existing queue one, if there) and when the startup is complete then we play that.
I know it's different to 1.0, but if you want it changed can you explain what you want, and please don't say "like 1.0" because there were problems with that. If you want it to not make any sounds whatsoever then thats something I can work with. I'll just add a DISABLE_ALL_BEEPS option if that's what you want.

Directory, 2021-11-11:
a disable all beeps option sounds great i would love that to be added before v2 is finalized. i can not express how much i lividly despise worthless feedback beeping. it should be made the default. and maybe remove disable_startup_beep since i dont think anyone is using it because they absolutly love and care that crsf connection is established.

JyeSmith, 2021-11-12:
Its actually a worth while feature to have as default.  ELRS covers a huge range of hardware and there have been issues with connections to the handsets.  I imagine we will have more questions with the introduction of MEGA BAUDs.  The beeps are a quick way for new users to know the module is receiving data.

FYI hot glue on the beeper also does wonders.

pkendall64, 2021-11-12:
Or just disable the buzzer in the FrSky header file!

wvarty, 2021-11-12:
I use both the CRSF connection beeps (super useful to debug connection issues) as well as the start-up tones.
@Directory I suspect you may be an outlier in your hatred for startup tones, and useful connection beeps. BLHeli is a perfect example of another FW in our hobby that uses this method to communicate state change, and look at how popular BLHeli32 jingle tones were / are.

Directory, 2021-11-12:
I love blheli startup tones cause bystanders get to hear "running from the 90s" from a drone followed by it going 99.9 ;) mph in a split second. That's cute. But im not picking up any girls with my obnoxious transmitter jingle coming out of the cheapest grape size speaker frsky was able to source. 

I also dislike the 2 "fc ready" beeps from my escs but I can't disable them and I won't because fc connection issues are a common point of failure often needing debugging by me on the field. Im not an elrs dev so idc about crsf ready because it would obviously work the majority of the time and always afterwards. Are you guys sure elrs users care so much about "crsf ready" as you think they do? Or is it just you guys since you're devs and like debug info.

JyeSmith, 2021-11-12:
Let me put it this way.  There were enough users issue to warrant adding the beeps to help trouble shoot.

wvarty, 2021-11-12:
@Directory nope, I have no idea if elrs users care so much about "crsf ready" as I think. Same as you have no idea either.
I have nothing against kindly and patiently requesting a "silent mode" flag.
It if far from important enough to be merged to 2.0, and will not be the default though.
You are welcome to fork the code and make it the default yourself if it is important to you 👍

CapnBry, 2021-11-30:
Fixed via #1098

closed 2021-11-30 by CapnBry