ExpressLRS/ExpressLRS #1090: LUA syntax error on Tango 2

wvarty, 2021-11-11:
## Current Behavior

When attempting to use the elrsV2.lua on TBS Tango 2, a syntax error is raised when spawning a popup. This occurs when attempting to trigger TX WIFI, or the BLE mode for example.

## Steps to Reproduce

1. Install elrsV2.lua on TBS Tango 2

2. Open LUA, and go to Wifi submenu

3. Attempt to trigger TX wifi

4. Syntax error

## Details

## Your Environment

- TX hardware: HM24TX slim pro
- RX hardware: NA
- Handset model: Tango 2
- OpenTX version (including nightly number): FTX
- ExpressLRS version (TX & RX MUST MATCH): 2.0RC1
- Packet Rate: NA
- Telemetry Ratio: NA
- user_defines: NA

kbzowski, 2021-11-11:
A similar error occurred on my Taranis X7 after upgrading to version 2
https://imgur.com/dveRbOC

WJL-FPV, 2021-11-11:
Using ELRSV2.lua with Tango2 + ES24TX-Lite, syntax error happened on WIFI submenu / BLE Joystick / Bind. Error message as attached photo (Tango2 screen is too small for the whole error message.

Tango2 firmware:
Tango2: V1.30
Tango2 XF: V6.07

Same ES24TX-Lite moved to X-lite (Running OpenTX 2.3.14) didn't get syntax error issue.

[image]

CapnBry, 2021-11-12:
Fixed via #1096

closed 2021-11-12 by CapnBry