ExpressLRS/ExpressLRS #1094: Dramatically decreased link quality in 50Hz mode

bede, 2021-11-11:
## Current Behavior

I recently tested 50Hz for the first time between two 150Hz flights in a rural location. Before testing 50Hz mode I flew a short flight in 150Hz with normal performance. After returning and switching to 50Hz mode, link performance was very poor and I started receiving LQ warnings at <300m range. These flights were back to back using the same battery. I would have tried 50Hz mode again, but it was getting dark.

In 50Hz mode, LQ had dropped into the 70s at just -70dBm. After I switched back to 150Hz I saw 100LQ below -100dBm as usual. I have DVR.

## Your Environment

- TX hardware: BetaFPV Nano module
- RX hardware: Happymodel EP1 with SDG regulator
- Handset model: Jumper T-Lite
- OpenTX version (including nightly number): 2.3.13
- ExpressLRS version (TX & RX MUST MATCH): 1.2.x-maintenance
- Packet Rate: 50Hz/150Hz
- Telemetry Ratio: 1:16
- user_defines:

CapnBry, 2021-11-12:
I can't reproduce this using an FM30 or a ES24TX and an EP2 receiver, but on 2.0-RC2.

Is it possible that the receiver was actually desynced and not fully connected? This would be evidenced by the LQ being <100 even when close.

I'd suggest a test where you set the TX power to 150Hz 10mW and walk until the LQ is around 90 (get a couple walls between you and the RX). Then switch back and forth and see how much difference the packet rate makes.

bede, 2021-11-12:
Thanks @CapnBry,
I'll try to reproduce this weekend and switch back and forth multiple times.
>Is it possible that the receiver was actually desynced and not fully connected? This would be evidenced by the LQ being <100 even when close.

I'll check DVR when home, but I think LQ was ok at close proximity.

bede, 2021-11-14:
I've repeated this test at the same location, and results were pretty consistent – 50Hz is performing considerably worse at the same power output in similar flights on my setup. For the 150Hz flight, LQ was 100 most of the time. For the 50Hz flight, LQ was highly variable throughout. 150, 250 and 500Hz all seem solid. Was using TrueRC moxon, as before.

From BF summary screen:

**50Hz**  
- min LQ: 71%
- min RSSI: -94dBm

**150Hz**
- min LQ: 95%
- min RSSI: -97dBm (flew further)

CapnBry, 2021-11-16:
I've done testing with two different receivers (ESP8285 and STM32) and two different transmitters (ESP32 and STM32) on 100mW 1:8 ratio and have the same results: 50Hz has slightly better LQ than 150Hz. This was on 2.0-RC2. Then I tested again on 1.2.0 with just the STM32 hardware at 100mW 1:16 150Hz and 50Hz and got the same results. I just can't reproduce your result at all and am getting the opposite result, which is in line with my expectation as well. 

You can't use a flight as a comparison when evaluating performance, there are far too many variables. 

The only thing I can think of in your case is that the continuous transmit time of 50Hz is 3x longer than 150Hz and maybe something in the RF hardware is fading or perhaps a crystal mismatch in hardware is causing the packets to not match up properly at the slower rate. However, I do not think there is any issue in ExpressLRS software.

bede, 2021-11-17:
Thanks for your efforts @CapnBry! Your hypothesis of a hardware issue is likely correct. I realise my testing isn't perfect, but the effect size is so large with my setup. I think I'll finally just buy the ES24TX-lite which seems to have a better reputation.

brandonrc, 2021-12-11:
Closing this issue, @bede if you have the same issue with the new ES24TX-lite let us know!

closed 2021-12-11 by brandonrc

bede, 2024-06-15:
Just FYI, I've not reproduced this issue with either of my other 2.4GHz Tx modules. Seems hardware specific