ExpressLRS/ExpressLRS #1095: build fails with "(" in HOME_WIFI_PASSWORD

jonas-ott, 2021-11-11:
## Current Behavior
Build of  2.0.0-RC1 fails if there's an `(` in `HOME_WIFI_PASSWORD`, with following error in the Configurator:
``sh: -c: line 1: syntax error near unexpected token `(' sh: -c: line 1: `xtensa-esp32-elf-g++ -o ".pio/build/HappyModel_ES24TX_2400_TX_via_WIFI/src/common.cpp.o"`` ...

[configurator-log.txt](https://github.com/ExpressLRS/ExpressLRS/files/7522583/configurator-log.txt)

## Steps to Reproduce

In ELRS Configurator:

1. Select 2.0.0-RC1

2. Enable HOME_WIFi

3. Use a wifi password with an `(` in it

4. Build

## Possible Solution (Not obligatory)

pkendall64 suggested on Discord to replace `(` with `\\\\050`, it fixed the issue

## Details

## Your Environment

- TX hardware: HM ES24TX
- ExpressLRS version (TX & RX MUST MATCH): 2.0.0-RC1
- OS: Manjaro Linux
- Configurator Version: v.1.2.3

pkendall64, 2021-11-11:
Added bug report to configurator https://github.com/ExpressLRS/ExpressLRS-Configurator/issues/130

JyeSmith, 2021-11-13:
@jonas-ott if you are comfortable testing PRs would you like to see if this is working for you. https://github.com/ExpressLRS/ExpressLRS/pull/1101

jonas-ott, 2021-11-13:
Yes, it builds and connects to my WiFi network.

CapnBry, 2021-11-23:
Closing since OP says resolved, re-open if that is not the case.

PR #1101 (2021-11-16): Perform shell escaping on build args
Depending on the characters used the build can failed or just get a corrupted SSID or password
An SSID or password with a special character '(' will do. So set the password to `fred(123`
The build will fail.

We need to escape the special characters. So far I have identified these as ones that need to be escaped.
I have found that if we replace them with 4 backslashes followed by the octal for character it works.
| Char | Replacement |
| - | - |
| ! | `\\\\041` |
| " | `\\\\042` |
| $ | `\\\\044` |
| & | `\\\\046` |
| ' | `\\\\047` |
| ( | `\\\\050` |
| ) | [... cut]
closed 2021-11-23 by CapnBry