ExpressLRS/ExpressLRS #1111: Telemetry lost during dynamic power test 2.0.0 RC2

unluckyfrog, 2021-11-17:
Elrs 2.0.0 -RC2
Edge tx 2.5.0 (28cbd40e) TX16s
Happy model 2.4 tx, EP1 rx.

250hz Paket rate
1:4 telemetry
TX Power dynamic on. Max 250mw

So I thought I'd try out the dynamic power setting on a flight today. Was working great, then for some reason I got a telemetry lost message and it got stuck on 10mw. I kept flying, as  Inav osd said I had 100LQ and ~85db. Telemetry reconnected after about 5 minutes, but lost connection soon after. 

I've since come home and done some "around the house" testing and it appears that changing the baud rate in edge tx to 921k fixes the problem. 400k, and 1.87k (which is what I was flying on) cause telemetry to drop out.

I don't really understand what Baud rate is, other than it having to do with communication between the radio and happymodel tx.

Wasn't sure whether to bring this up in elrs or edgetx github. Apologies if this is the wrong group.

SunjunKim, 2021-11-17:
Do you set AUX1 as the arm switch, and set "high" as an arming?
With RC2 code it's expected to jump up to the max power when telemetry is lost.

unluckyfrog, 2021-11-17:
Yes Aux 1 is arm switch.  
hmmm that's embarrassing, I have arming set as low. I did't check this ><.
THX

SunjunKim, 2021-11-17:
It's not uncommon to have telemetry lost, as downlink is much weaker than uplink on EP1.
If you often encounter challenging situations, consider upgrading your RX to one has PA/LNA.
https://www.expresslrs.org/2.0/hardware/receiver-selection/

avsaase, 2021-11-17:
> Yes Aux 1 is arm switch. Arming on low, not high though. Didn't think that matters. Can change though.

Yes, this is important for dynamic power to work correctly. When you lose telemetry it will increase power to the max configured level, but only when the arming channel is in the high position. This is to avoid blasting at full power when switching batteries.

unluckyfrog, 2021-11-17:
**avsaase** "This is to avoid blasting at full power when switching batteries" 

Ahhh that makes sense, very clever. 
Thanks

unluckyfrog, 2021-11-17:
**Sunjunkim** "consider upgrading your RX to one has PA/LNA"
Done and ordered. Thanks for the advice.

closed 2021-11-24 by unluckyfrog

AerialClone, 2021-12-28:
I have the arming in low and I made the set up long ago and I can't remember how to change it to high... anyone can help?

jurgelenas, 2021-12-28:
> I have the arming in low and I made the set up long ago and I can't remember how to change it to high... anyone can help?

It is configured through OpenTX/EdgeTX. Go to model settings -> Mixes tab and set CH5 (AUX1) weight to -100. Do not forget to update betaflight arming as well. Something like this: 

[image]

For the future: it is unlikely to get answers for questions like this in GitHub issues. Please use ExpressLRS discord help and support channel :) there are lots of smart people willing to help you.

Vittaly, 2023-04-04:
Hi All!
> With RC2 code it's expected to jump up to the max power when telemetry is lost.
I have FC with ArduPilot and arming is makes through throttle stick. 5th channel is using for other function. Is it mean thet I can not use function "dynamic power"?