ExpressLRS/ExpressLRS #1124: TX fails to go into WiFi update mode. (2.0.0-rc2)

sherghan, 2021-11-21:
## Current Behavior
BetaFPV Nano TX (2.4Ghz) fails to go into WiFi update mode.

## Steps to Reproduce

1. Turn the radio on & switch to model controlled by ELRS external module

2. Execute the ELRS Lua script (Version compatible with 2.0)

3. Go into _[WiFi Connectivity]_

4. Choose _[Enable WiFi]_ and accept (RX is not connected)

For all the details please check the following video:  
[youtu.be/ooSAoECMawY](https://youtu.be/ooSAoECMawY)

## Possible Solution (Not obligatory)
I suspect this might be connected to  #1082 ...common solution?

## Details
Please ask for any additional details that might help.
I'm willing to test possible solutions.

## Your Environment

- TX hardware: _BetaFPV 2.4Ghz Nano TX_
- RX hardware: _not relevant_
- Handset model: _Jumper T-light_
- OpenTX version: _2.3.14-otx (1e09791a)_
- ExpressLRS version (TX & RX MUST MATCH): _2.0.0-RC2 (a8d405)_
- Packet Rate: _250_
- Telemetry Ratio: _1:16_
- user_defines: _binding_phrase, uart_inverted, feature_opentx_sync, auto_wifi_on_interval=90_

brandonrc, 2021-11-22:
Same issue on the BetaFPV 2.4ghz Micro TX

JyeSmith, 2021-11-23:
The BETAFPV Nano 2.4 (I haven't explicitly tested the others) reboots when starting WiFi or Bluetooth.

The target is exactly the same as the DIY 2.4 TX with the addition of some LEDs and a button.  Commenting these out of the target does not fix the issue, nor should it actually be the problem.

```
#define GPIO_PIN_LED_BLUE       17
#define GPIO_PIN_LED_GREEN      16
#define GPIO_PIN_BUTTON         25
```

In the end I don't know why and the ESP32 datasheet doesn't indicate an issue with the pins and their use.

Im just going to throw it out there that it is a possible hardware issue.

kimman1, 2021-11-23:
With my DIY 2.4 TX. The exact same problem. The way I try to fix is turn the radio off and on again and the problem solve and I have another problem 
Go to Wifi conectivity folder on Lua Scripts and choose enable RX wifi, Lua show Sending... and disappear but RX can't active wifi

drewgarrido, 2021-12-31:
There is a known issue with the BetaFPV Nano TX not going into WiFi update mode or being flashed via USB. They missed a capacitor on enable line of the ESP32 chip for the "first batch" of production.

https://www.youtube.com/watch?v=Pt5rMp7Zs8s

sherghan, 2022-01-01:
This video is about USB flashing problems that I don't have with my module. I always flash it via USB without any problems.

It mentions WiFi firmware update problem, but it never states that the capacitor solution, works for this too.
Additionally I don't think I have a module from the "first batch".

I'm still unable to use BT nor WiFi in this module. :-/

brandonrc, 2022-01-07:
https://github.com/ExpressLRS/ExpressLRS/issues/1082

This is the same issue. Closed 1082 with this statement:

Regulator is under spec and the module browns out.

Please contact BetaFPV or the company that you purchased the product from.

closed 2022-01-07 by brandonrc