ExpressLRS/ExpressLRS #1125: R9m 2018 + tx12 "FW update error Not responding"

fernandop009, 2021-11-21:
## Current Behavior
Hi everyone,

I have a problem trying to install the bootloader "r9m_elrs_bl".The error say: " FW update error Not responding". My tx is a Radiomaster tx12 (with opentx 2.3.14) and my module is the R9M 2018 (with the resistor mod done).

Does anybody have this issue? 

Thanks for the great work.

## Your Environment

- TX hardware: radiomaster tx12
- RX hardware: r9mm
- Handset model:
- OpenTX version (including nightly number): 2.3.14
- ExpressLRS version (TX & RX MUST MATCH): 1.2.0
- Packet Rate:
- Telemetry Ratio:
- user_defines
[image]
:

fernandop009, 2021-11-22:
Finally, I found the problem. when I did the "r9m inverted mod " the resistor break me the PCB pad from the inverter chip. the module still works but I cannot upgrade any firmware. so be careful if you need to do this mod

closed 2021-11-22 by fernandop009

EinApfel1, 2022-05-14:
Thank you for the information were you able to soldering it back on ? I ran into the same problem :/ guess i have to order the 2019 r9m to no have to do the mod again