ExpressLRS/ExpressLRS #1126: RX not reconnecting to TX after connection loss.

brandonrc, 2021-11-23:
## Current Behavior

https://discord.com/channels/596350022191415318/798006228450017290/912518060903129088

RXLOSS on ELRS 500mW 2.4ghz BetaFPV Micro TX. 

During a flight the connection dropped and put the drone into RXLOSS. During RTH the RX did not reconnect. 

The user did not enable NO_SYNC_ON_ARM so that did not cause the issue. 

https://youtu.be/Sybym591iks

RX did not reconnect when the drone got closer. 

## Steps to Reproduce

1. Fly the quad until you lose signal 
2. The RX will not get the signal back. 

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware: BetaFPV Micro TX (Subwoofer)
- RX hardware: EP1

JohnDGodwin, 2021-11-23:
Pilot here. Trying to add any relevant information.
I fly DJI mostly and this is one of my first analog flights, so the OSD is not setup properly. I have LQ channel mapped.

Packet rate was 250hz, telemetry ratio 1:64

brandonrc, 2021-11-23:
I could not repeat this issue.. 

@JohnDGodwin try and see if you can repeat this.. like use low power.. and try and fail safe..

estebanavv, 2021-11-23:
@brandonrc I suspect that you have an issue in your TX, right before the RX loss it was briefly alternating between 99 and 0 then when is landing it flashes twice 99.  I had a similar issue where the board was separating from the case and it had a faulty contact.

brandonrc, 2021-11-23:
> @brandonrc I suspect that you have an issue in your TX, right before the RX loss it was briefly alternating between 99 and 0 then when is landing it flashes twice 99. I had a similar issue where the board was separating from the case and it had a faulty contact.

This was not me. I was just writing up a ticket to help someone. 

@estebanavv do you have the new betaFPV tx with oled?

JohnDGodwin, 2021-11-23:
@estebanavv Which TX module? Do you know where specifically the faulty contact was?

estebanavv, 2021-11-23:
@JohnDGodwin This happened to me with the original BetaFPV TX. Yes the faulty contact was because although the case was firmly pressed inside the transmitter, the board was lose and was pushed inside the case so the contacts failed where the board connects to the transmitter. To fix it I used screws to firmly attach the board to the inside of the case.

JohnDGodwin, 2021-11-23:
> 

I disassembled my tx and inspected it. It is very solid with no wiggle room. I do not think this is the issue.

estebanavv, 2021-11-23:
Based on the video and the signal level shown on the OSD this doesn't seem to be a firmware issue, unless this can be reproduced it should be closed.

brandonrc, 2021-11-27:
Closing issue until it shows back up.

closed 2021-11-27 by brandonrc

JohnDGodwin, 2021-12-04:
I just wanted to clarify what happened.
I'm 99% sure the tx and rx wires on the receiver came in contact with one another during flight.
The issue of being unable to take back control was user error. I can confirm this because after the quad lands, I activated dshot beacon and the beacon sounded, which means I had connection again.

johndoe422, 2024-01-02:
I have a Radiomaster TX12 MKII transmitter running 3.3.0, and an RP1 receiver. I can confirm that this issue is present in my setup. Once the radio connection is lost. it never re-establishes the connection until the receiver is power cycled. The receiver's green LED will be blinking and not solidly lit up until that.
Packet Rate: D250 (-104 dbm)
Telem Ratio:  Std (1:128)
Switch Mode: Wide
Model Match : Off (ID:0)
Tx Power: 250 mW