ExpressLRS/ExpressLRS #1132: Enable RX WIFI from the LUA Script does not always work as intended

Pelado-Mat, 2021-11-23:
# Current Behavior

## 500HZ 1/64 TELM - RX: HM EP1 or EP2 ##
Most of the times, I get this:
1. Turned on TX/Radio
2. Plug LiPo in.. 
3. TX and RX connects.. fine
4. Enable RX WIFI - Nothing Happens  WIFI mode does not activate
5. Power Cycle - RX enters in WIFI mode at once

## 150HZ 1/16 TELM - RX: BetaFPV RX ##
Most of the times, I get this:
1. Turned on TX/Radio
2. Plug LiPo in.. 
3. TX and RX connects.. fine
4. Enable RX WIFI - Works fine

## 500HZ 1/64 TELM - RX: BetaFPV RX ##
Most of the times, I get this:
1. Turned on TX/Radio
2. Plug LiPo in.. 
3. TX and RX connects.. fine
4. Enable RX WIFI - Nothing Happens  WIFI mode does not activate
5. Power Cycle - RX enters in WIFI mode at once

# Environment
## HW:

1. TX HM ES2.4TX
2. RX HM EP1/EP2 BetaFPV RX

## SW:

1. version 2.0RC3
2. Model Match: ON
3. Dynamic Power: ON

CapnBry, 2021-12-01:
I can confirm that there are a lot of cases where the RX does not get the MSP command to go into Wifi mode. The MSP subborn sender is getting jammed up. I have an idea why this may be happening but will look into it. Thanks for reporting.

CapnBry, 2021-12-01:
Can you try using the Configurator and flashing Git Pull Request -> #1145  and see if this resolves the issue for you?

Pelado-Mat, 2021-12-02:
> Can you try using the Configurator and flashing Git Pull Request -> #1145 and see if this resolves the issue for you?

I have just tested the fix, It's working! I tried with an ep2 rx in different packet rates and telemetry ratios, all worked flawlessly.

congrats!

closed 2021-12-06 by Pelado-Mat

Gerd-St, 2022-05-17:
Hi, could it be the case that this issue has ben fixed for 2.4GHz devices only?
I have several BetaFpv 900MHz receivers which I have flashed with AUTO_WIFI_ON_INTERVAL disabled. When I then try to switch them to WIFI mode via LUA script nothing happens, regardless of what tx settings have been applied before. Also no switching to WIFI mode after power cycling.
I tried both with firmware versions 2.3 and 2.5.

bryanmorganoverbey, 2023-03-18:
Same as Gerd-St, I could connect to the Happymodel EP1 but could not get the lua script to activate wifi mode. Turned out the channel for ARM mode was ON, so the lua script ignored the command to enter wifi mode

psfales, 2023-10-19:
> 
> PROBLEM SOLVED I found out that the LUA command gets ignored while CH5 is high (2ms). I had set this to always armed, as I have no use for it with my servo based applicaiton.

Thanks for providing the solution!   In my servo-based application, I had another channel mapped to the channel 5 output pin, and wasn't thinking about channel 5 (Aux1) at all!