ExpressLRS/ExpressLRS #1134: `src/python/build_flags.py` breaks the build process if `pip` is not available

wucke13, 2021-11-23:
I don't use `pip`, thus I don't have it installed on my system. Hence [the code in `src/python/build_flags.py`](https://github.com/ExpressLRS/ExpressLRS/blob/a3c9f47e493e0510444304c47b5ec292cfc57fd4/src/python/build_flags.py#L75) doesn't work on my system. Further on I think it's really not nice to try to mutate the system state by running `pip` without user consent, but more on that later.

When I run `platformio run -t upload -e <myenv>`, I get the following error

```
/nix/store/5bh6rpya1ar6l49vrhx1rg58dsa42906-python3-3.9.6/bin/python3.9: No module named pip
CalledProcessError: Command '['/nix/store/5bh6rpya1ar6l49vrhx1rg58dsa42906-python3-3.9.6/bin/python3.9', '-m', 'pip', 'install', 'GitPython']' returned non-zero exit status 1.:
  File "/nix/store/h30rry7pkfpgvy3y0m2qpincsjdxl7da-platformio-5.2.3/lib/python3.9/site-packages/platformio/builder/main.py", line 178:
    env.SConscript(item, exports="env")
  File "/home/wucke13/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/wucke13/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/wucke13/documents/projects/c/ExpressLRS/src/python/build_flags.py", line 110:
    build_flags.append("-DLATEST_COMMIT=" + get_git_sha())
  File "/home/wucke13/documents/projects/c/ExpressLRS/src/python/build_flags.py", line 74:
    subprocess.check_call([sys.executable, "-m", "pip", "install", "GitPython"])
  File "/nix/store/5bh6rpya1ar6l49vrhx1rg58dsa42906-python3-3.9.6/lib/python3.9/subprocess.py", line 373:
    raise CalledProcessError(retcode, cmd)
```

which indicates that the cascaded handling of failure on `import git` does not consider the possibility that `pip` is not installed (despite trying so by using multiple ways of calling `pip`). I would propose the following variant of the `get_git_sha` function.

It solve two issues: first, it does account for the possibility that `pip` is not failing but rather just not installed. Second it tries yet another robust way of getting the sha-1, by simply directly calling `git` the binary (which at least every slightly development related lad likely has in their `$PATH`). 

If it is allowable, of course, I would much prefer not trying to call `pip` at all. IMHO its the wrong thing to do, for one, its not necessarily the way the user wants to handle python dependencies, second I don't want some build-script to alter my system state without my consent and third, I don't believe its too much to ask the user to either install `git` or `python-git` to get things running. But this is of course up to opinion. Following my proposal of the function:

```python
def get_git_sha():
    # Don't try to pull the git revision when doing tests, as
    # `pio remote test` doesn't copy the entire repository, just the files
    if env['PIOPLATFORM'] == "native":
        return "0x00,0x11,0x22,0x33,0x44,0x55"

    git = None
    try:
        import git
    except:
        try:
            sys.stdout.write("Installing GitPython")
            subprocess.check_call([sys.executable, "-m", "pip", "install", "GitPython"])
            import git
        except:
            try:
                env.Execute("$PYTHONEXE -m pip install GitPython")
                import git
            except:
                pass

    sha = None
    if git:
        try:
            git_repo = git.Repo(
                os.path.abspath(os.path.join(os.getcwd(), os.pardir)),
                search_parent_directories=False)
            git_root = git_repo.git.rev_parse("--show-toplevel")
            ExLRS_Repo = git.Repo(git_root)
            sha = ExLRS_Repo.head.object.hexsha
        except git.InvalidGitRepositoryError:
            pass
    else:
        sha = subprocess.check_output([ "git", "rev-parse", "HEAD" ]).decode("utf-8").rstrip('\r\n')
        #raise Exception(sha)

    if not sha:
        if os.path.exists("VERSION"):
            with open("VERSION") as _f:
                data = _f.readline()
                _f.close()
            sha = data.split()[1].strip()
        else:
            sha = "000000"
    return ",".join(["0x%s" % x for x in sha[:6]])

```

wucke13, 2021-12-23:
My proposed fix doesn't work anymore, since the code moved to `python/elrs_helpers.py`. The issues remains. It's kind of frustrating to see that somebody refactored the code without addressing existing issues. In hope that this gets some attention: @AlessandroAU .

pkendall64, 2021-12-23:
Could you make a PR and we'll review and get it merged. Target it for master branch.

wucke13, 2021-12-23:
Yes, I can look into that. However, this just got harder since now it looks like we can't get away without the python git module.

pkendall64, 2021-12-23:
That is true, because we use the tag/branch name and commit hash as the version information in both LUA and Web UIs.

wucke13, 2021-12-23:
#1214 should address this issue in partial, let's move the conversation over to there?

commit b1cda25c (2021-12-29): fix #1134: don't require python git module (#1214)
closed 2021-12-29 by pkendall64