ExpressLRS/ExpressLRS #1136: ELRS 2.0, show regulatory domain in lua

error414, 2021-11-23:
Hi,

would be nice if regulatory domain was be visible in lua script. I use it for double check which FW I flashed to module. In europe could be big problem to transmitting at 900Mhz, so better to check it again. 

It can be side by version of FW, for example 

2.0.0 EU_868

CapnBry, 2021-11-30:
We probably could build that into the version number, great idea! I'll look at this for 2.1.

I'm not sure how much of a problem this is considering nobody should ever be changing their domain, but is worth the few bytes it adds.

AlessandroAU, 2021-12-01:
Ironically this is a regression since we had it displayed in V1.X LUA... oops.

CapnBry, 2021-12-01:
I think we just didn't want to waste the flash space and Lua load time to show it, but just adding 4 bytes (` 868`, ` 2G4`, etc) to the existing version hash is a really great idea since it still fits on all screens and doesn't add much. Just need to update the build process to tack it on.

closed 2022-01-07 by maybenikhil