ExpressLRS/ExpressLRS #1137: Tracer TX flashing?

FiercestT, 2021-11-24:
Not sure if anyone has asked this or if there are some differences, but would it be possible to flash ELRS to a tracer TX and get that juicy power output? Like how you can flash ghost or an R9 to ELRS.

From my understanding, tracer supports FLRC + LoRa (with the announced update).

Is this a matter of someone making a firmware target/guide or is there more to it?

brandonrc, 2021-11-25:
@FiercestT so the issue is not the radio chip used, sx1280 which we support. The issue is the MCU TBS uses which is the PIC32. 

So I can list a few of the quick highlights during my adventures of PIC32 development. 

Cons of bringing PIC32 support to expressLRS:

1. PIC32 is owned by microchip and they like to sell compilers and support. (Not good for opensource)
2. The opensource community does not use PIC32 at all.. (1 or 2 projects like fubarino and chipkit)
3. Flashing requires a custom flashing tool that cost around $30 to $80 usd.
4. The clock is half the speed as the average STM and a third or or less the speed of ESP. 

Pros:
1. People can use Tracer with ELRS
2. Trappys face... 

I have spent over 100 dollars on PIC32 development gear and hope that one day I can get ELRS working for the sake of working but I am hesitant to follow throw with this support. I think this is something that might have to be released for fun but not maintained by ELRS devs.

brandonrc, 2021-11-25:
https://github.com/ExpressLRS/ExpressLRS/issues/645

closed 2021-11-25 by brandonrc