ExpressLRS/ExpressLRS #1140: Lua script syntax error (2.0.0 RC3 lua)

charlesmarseille, 2021-11-27:
## Current Behavior

Entering Lua script takes a long time (~10 minutes until full  menu) then warning popup with message:
Script syntax error
bad argument #1 to 'char' (number expected, got nil)

## Steps to Reproduce
Open EdgeTx controller, start lua script.

## Your Environment

Tx16s, edgetx 2.5.0, elrs 2.4 2.0.0 RC2, diy e28 tx module

RipperGOT, 2021-11-28:
Tx16s, edgetx 2.5.0, elrs 2.4 2.0.0 RC3, HM24tx module, no problem.

CapnBry, 2021-11-30:
Have you checked the baud rate in the radio settings to verify it is 400k? I have a TX16S and it worked with every module I have (FM30, Namimno OLED, ES900TX).

jurgelenas, 2021-12-11:
@charlesmarseille were you able to figure this one out?

nwpulver, 2021-12-14:
same issue on access qx7 (with inverter mod done) on edge tx. my error says `syntax error: elrsv2: 491 "}" expected` . I think it is a curly brace, hard to tell

schugabe, 2022-01-21:
@charlesmarseille could you test if elrs 2.1 still has that issue on your radio (update lua script + modules)?

charlesmarseille, 2022-01-29:
Sorry for the delay, I confirm the bug is gone!

closed 2022-01-29 by charlesmarseille