ExpressLRS/ExpressLRS #1141: FrSky R9 TX communication problem [2.0.0-RC3]

RomkaV, 2021-11-28:
After upgrading from version 1.2 to 2.0.0-RC3, it is not possible to configure FrsSky P9. Settings are not saved.

[Steps to Reproduce](https://youtu.be/2nc4--IKlLA)

Transmitter: Taranis QX7 ACCT, EdgeTX 2.5.0
TX Module: FrsKy R9M

[image]

CapnBry, 2021-11-30:
I believe this should be fixed by the latest 2.0.x-maintenance branch and when 2.0.0-RC4 comes out.
https://github.com/ExpressLRS/ExpressLRS/pull/1133

Can you try using Branch -> 2.0.x-maintenance in the configurator to verify? Only the TX needs to be flashed.

hyperion11, 2021-12-03:
use R9M 2019 with OpenTX 2.3.14 2.0.0-RC3. All settings saved except WideChannels.

CapnBry, 2021-12-03:
> use R9M 2019 with OpenTX 2.3.14 2.0.0-RC3. All settings saved except WideChannels.

Great!

You can't change switch mode with a receiver connected, that's be design.

RomkaV, 2021-12-05:
Thanks! Solved!

closed 2021-12-05 by RomkaV